MarlinFirmware/Marlin #13198: Thermal Runaway Not Caught

nionio6915, 2019-02-18:
Thermal Runaway Not Caught

I would advise against trying to reproduce this- install a 12V heater cartridge powered by 24V controlled via a mosfet with trigger wires on RAMPS D8 pin. Failure of the mosfet results in stuck on condition. 
  
**Expected behavior:** Marlin to abort after MAXTEMP threshold exceeded or thermal runaway detected.

**Actual behavior:** Heater is stuck on and nothing triggered or failsafe condition met to shut system down. 

#### Additional Information
#define HEATER_0_MAXTEMP 285

#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius

https://i.imgur.com/t47Mpoj.png

[image]

Voltage 'spikes' are me killing the power trying to figure out what is going on. As you can see, the heating starts right back up as the mosfet is cooked. 

[configs_thermal_runaway.zip](https://github.com/MarlinFirmware/Marlin/files/2874586/configs_thermal_runaway.zip)

AnHardt, 2019-02-18:
Try a more recent version of Marlin. 1.1.7 is likely to be before my last fix of that.

RudyBenner, 2019-02-18:
How is this a firmware issue?

marcio-ao, 2019-02-18:
@nionio6915: A stuck MOSFET is one of the cases in which there is nothing Marlin can do. Basically of the MOSTFET is stuck on, then it simply won't respond to Marlin's request to turn it off. End of story.

The only way to protect against this is to modify your hardware. Possibilities:

- Add a themal fuse to the hotend that will cut power to the hotend if the temperature exceeds a maximum.
- Add a relay to the power supply and wire it up to the board so that Marlin can cut power to the printer if it detects a runaway and the MOSFET is not responding.

AnHardt, 2019-02-18:
He should at least see the MAXTEMP_ERROR with kill. If configured we can switch off the power supply.

marcio-ao, 2019-02-18:
@AnHardt: Ah. Gotcha.

nionio6915, 2019-02-19:
marcio-ao:  you're right, the mosfet once runaway is not stoppable, BUT I expected to have some sort of thermal runaway message. I did not realize what was happening until I smelled the perfume of heating plastic, then only looking at the display did i realize what happened. 

In hindsight, I am sorry I did not think to pull the Octoprint logs or buffers. I pulled the psu's power chord not thinking more than stopping a fire. 

AnHardt: Yes, that is what I was expecting, some sort of function triggered by the overtemp. I thought I was more current than 1.1.7. I will migrate to more current version- now is as good as any reason to change.  Can you point me to what section in the configs or scripts to look at to figure this out more?

Octoprint has Temp Failsafe plugin - and I can have it command a script if the temp is exceeded. I am not sure I am completely confident in it alone.

AnHardt, 2019-02-19:
From Your Configuration.h:
> #define CONFIGURATION_H_VERSION 010107

boelle, 2019-02-20:
is this one solved?

AnHardt, 2019-02-20:
@thinkyhead 
> * Did the machine simply lock up and become unresponsive?

Take a look at the firs message. The graph can not continue when killed.

thinkyhead, 2019-02-20:
> Take a look at the firs message. The graph can not continue when killed.

tl;dr

None of this is very clear. After reading it several times, I think it says:
- Your MOSFET fails for some reason and gets stuck on
- Marlin isn't monitoring the heaters so it fails to catch the error

So it sounds like all we need is to monitor the heaters at all times.

AnHardt, 2019-02-20:
I'm quite sure we fixed a bug in that area last year. His Configuration.h tells us he is on 1.1.7, what is a lot older. All we have to do is waiting for @nionio6915 to confirm it is fixed. Only if he can't confirm, we have to have an other look into it.
https://github.com/MarlinFirmware/Marlin/issues/13198#issuecomment-464656406
Why do we have a that extended discussion about that a simple thing?

@thinkyhead 
Your new tests should already be done in the temp-irq against the raw values.

thinkyhead, 2019-02-20:
Yes, I checked it thoroughly, and everything is in good order in the latest Marlin.

closed 2019-02-20 by thinkyhead

nionio6915, 2019-02-21:
Sorry for my late response- i have been away.  To clarify - 
I am using 8 bit RAMPS 1.4 / Mega 2560, so I am limited to 1.1.9.   I thought I upgraded to 1.1.9 (I opened an issue about a typo in 1.1.9 referencing 2.0). Regardless, I will upgrade to latest version and reconfigure. 

1. I believed that HAD enabled Thermal Runaway Protection in the configs. I had gotten thermal runaway halts for not reaching temps before. I also had a false sense of security because on one printer with older Repetier FW, in Octoprint I get a warning about not having thermal runaway protections. On this printer with Marlin I don't get that warning. Which just goes to show you that the lack of warning does not mean anything if its not configured properly. Shame on me for not thinking it through more.  

2. The behavior that I saw was that no warning or action occurred in the overtemp condition. In the undertemp condition there is a system halt and system reset. as the max_heater temp was exceeded I was expecting something to happen. Nothing happened on my machine even after a period of more than 5 minutes over the limit of 285C. 

3. My understanding is that over the period of time  (THERMAL_PROTECTION_PERIOD) Marlin would compare temps, and if it was over the threshold (THERMAL_PROTECTION_HYSTERESIS), then Marlin would trigger the thermal runaway condition. If I am mistaken, please point towards the place to read and learn it better.

4. Not a marlin issue - I did not realize the disconnect I created is that the by passing off PSU control to Octoprint via PSU control plug causes another conflict.  Any control of the PSU now has to be done on the raspi. i am not sure yet how to reconfigure my setup with my hardware to create the best setup.

I will upgrade Marlin and attempt to re-test the scenario by unplugging the heater and wiring in a potentiometer to simulate thermistor input. I still have the defective mosfet but think its best to try a 'controlled' scenario.

Thanks again for all your input.

boelle, 2019-02-21:
> I am using 8 bit RAMPS 1.4 / Mega 2560, so I am limited to 1.1.9

not true, bugfix 2.0 is also 8 bit