MarlinFirmware/Marlin #13202: [bugfix 2.0] Bltouch and G28 problem on board with separate Z_MIN_PROBE

StevenP94, 2019-02-18:
### Description
I use a RuRamps4D v.1.3 that has commented Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and uses Z_MIN_PROBE_ENDSTOP instead, and a BLTOUCH.

### Steps to Reproduce
1. Configure Marlin to use Z_MIN_PROBE_ENDSTOP instead of Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

2. Configure Marlin for BLTOUCH

3. Try to do a G28

**Expected behavior:**
X and Y homing using endstops, Z uses bltouch to set Z home.

**Actual behavior:** 
X and Y works fine, bltouch not deployed, nozzle crashes on bed.

#### Additional Information
All works fine, M43 S command uses bltouch without problems.

**#### RESOLUTION**
Go to Conditionals_LCD.h at row 519 and change it from

`#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
`
to 

`#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && (ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) || ENABLED(Z_MIN_PROBE_ENDSTOP)))
`
This enable the use of bltouch.
 I'm sorry to abuse of Issues but I don't know how to make a patch request, so I hope that  this can be useful.

I Apologize for my english

boelle, 2019-02-18:
does your fix work when homing is done from pc and lcd?

StevenP94, 2019-02-18:
yes

boelle, 2019-02-20:
maybe submit a PR?

boelle, 2019-02-20:
maybe close this ?

thinkyhead, 2019-02-20:
The existing behavior is by design. If you have a real Z endstop, then it will be used for Z homing. If you don't have a real Z endstop, then you should plug your probe into the Z min endstop plug. If you can't plug the probe into the Z min endstop plug, then you should just change your `Z_MIN_PIN` to the pin to which your probe is connected.

closed 2019-02-20 by thinkyhead

StevenP94, 2019-02-20:
> The existing behavior is by design. If you have a real Z endstop, then it will be used for Z homing. If you don't have a real Z endstop (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN), then you should plug your probe into the Z min endstop plug. If you can't plug the probe into the Z min endstop plug, then you should just change your Z_MIN_PIN to the pin to which your probe is connected.

I don't understand how can it work making a test on Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN only that, as you said, is disabled because I use a real Z endstop. My change just enabled the bltouch when a real Z_MIN_PROBE_ENDSTOP is enabled.

thinkyhead, 2019-03-01:
`Z_MIN_PROBE_ENDSTOP` has been removed. The only option now is to enable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` or to not enable it. It might be renamed to something like `Z_PROBE_IS_Z_ENDSTOP` in future, if this better captures the full meaning of the option.

boelle, 2019-03-01:
will there also be an option to spec what pin to use as probe? just to support those boards with seperate probe connector?

AnHardt, 2019-03-01:
@thinkyhead 
That would express better the implications:
* There is no z-min-endstop 
* Use the probe for homing to z-min.

But non't ask me for names: I already suggested too many of them in that complex. All reverted or ignored. :-(

thinkyhead, 2019-03-03:
> Will there also be an option to spec what pin to use as probe?

Yeah, that was done too.