MarlinFirmware/Marlin #13203: past 1000 open issues, it can't be right

boelle, 2019-02-18:
we have yet again passed 1K open issues

This can't be right, should they not be reviewed and labeled so things are not lost ?

Maybe close those where people have lost interest etc, ie a good general cleanup

i know that the list can be sorted by what is updated most, but still 1000 open issues?

InsanityAutomation, 2019-02-18:
And you made another issue to discuss too many issues instead of joining us on Discord or the Facebook Group something?

Sorry, just couldn't resist laughing at the irony :)

For useful content here though, in the past ive used Redmine where issues can be tagged to versions, or target milestones for tasks and feature requests ect and it integrates with Git nicely. My recommendation would be to implement something more easily manageable.

boelle, 2019-02-18:
it was more for the "food for thought" factor

i used to be on facebook but have given it up, its just not for me
and discord suffered the same ill faith

and going there would not change the "problem" here :-D

InsanityAutomation, 2019-02-18:
@boelle Understandable but im sure you can see the irony I had to laugh about :) A large number of the issues are gone, just users dont ever come back and close them. I try to go back and remind them to on stuff ive fixed, and to confirm it actually fixed it! Not everybody ever comes back though...

boelle, 2019-02-18:
hehe

i guess that is about or at least 25% of the problem, my thought was to start at the very bottom of the list and ask people if they still have a problem and if not close the issue, or if no reply within a week also close it with a remark that they can post a new issue

InsanityAutomation, 2019-02-18:
Would need to do so while also looking for dupe's and point users to the oldest issue on a topic and close newer, with comments stating issue # to prevent lost discussion. Time consuming task though.

boelle, 2019-02-18:
yep... and they have made a long list of labels with colors etc

but given i dont have the "powers" to do anything other than creating and closing my own issues all i could do was raising the question if it was not time to do "something" before it all snowballs to 2000 open issues :-D

swilkens, 2019-02-18:
You seem motivated to dig through them all and make a list of "can be closed" / "stale" issues that somebody with admin status can dig through in a more efficient way.

I took a peek a while ago and found many of them are users asking questions that should be redirected to facebook or discord. That alone will take care of ~200 issue reports.

boelle, 2019-02-18:
i could do so but i dont want to pi** anyone off

if there was just an access level that allowed me to add the labels that would help a lot, and at same time i could ask if the issue was still relevant

@thinkyhead thoughts?

boelle, 2019-02-18:
do we have an "official" discord? the facebook group i know

but i want to have some sort of green light before i start to direct people towards those places

swilkens, 2019-02-18:
> i could do so but i dont want to pi** anyone off
> 
> if there was just an access level that allowed me to add the labels that would help a lot, and at same time i could ask if the issue was still relevant
> 
> @thinkyhead thoughts?

It's a community effort, janitorial work is always usefull and there's nothing wrong with guiding people to a place better suited for their questions.

I would propose you dig through all the issues, find the ones that seem obviously out of place / stale / irrelevant and make a clickable list - post that here and I'm sure your efforts will be valued :)

InsanityAutomation, 2019-02-18:
> do we have an "official" discord? the facebook group i know
> 
> but i want to have some sort of green light before i start to direct people towards those places

Yeah, a search for discord in the issues section will turn up the link, only mentioned a few times though. A good number of people still find it!

boelle, 2019-02-18:
i agree, we could even come up with a standard template of text to insert?

boelle, 2019-02-18:
what about this text:

Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/ 
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

closed 2019-02-18 by thinkyhead

Bob-the-Kuhn, 2019-02-20:
I've been trying to find examples of people getting non-trivial support on Facebook and Discord.  So far I've only noticed examples that were 1 day long.  

Could someone do me a favor and point me to an example in each medium?

DerAndere1, 2019-03-05:
@boelle I'm sure that by contacting reporters of outdated but open issues you will help a lot. However, Your template text in https://github.com/MarlinFirmware/Marlin/issues/13203#issuecomment-464865468 probably contains a mistake. A better (corrected) template for redirecting users is given below. 
I guess you wanted to redirected users to the **official, open** MarlinFirmware facebook group. But keep in mind that some issues are open for a long time simply because there exists no (simple one) solution. One problem with facebook is: it has no standarized tags like "closed" / "solved" , so experts cannot use facebook's search to dig out content that was posted some days ago and filter for posts that contain unsolved issues, so they cannot provide help where it is needed most. That is why facebook is usually not the right place for discussing complex! issues. Nevertheless, here is a corrected template:

Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/1049718498464482/ 
The issue list is for bugs and feature requests only.
Please log in to your account at https://github.com and close this issue once you have posted it on one of the 2 links.
Thanks!

thinkyhead, 2019-03-05:
This is the boilerplate text I use as the starting-point for diverting support requests to other forums. Sometimes posted as you see here, but typically edited down depending on the situation.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

---