MarlinFirmware/Marlin #13204: [RFC, DUE] Call HAL_idletask() from minkill()

marcio-ao, 2019-02-18:
Right now, if the machine enters the kill() handler, the serial messages don't often get written and the serial port locks up. I suspect this is because on the DUE, the serial code requires HAL_idletask() to be called.

Would it make sense to call HAL_idletask() from within minkill() to keep the serial port going?

marcio-ao, 2019-02-18:
Also, it seems like some fairly long commands, like G29, don't call the idle task either. It seems to me that HAL_idletask() should be called from manage_heater()?

marcio-ao, 2019-02-18:
Closing this, since I am not sure now whether this is an issue. Investigating further.

AnHardt, 2019-02-19:
> Would it make sense to call HAL_idletask() from within minkill() to keep the serial port going?

We could wait a bit longer, until all messages are out (if that helps).
After that we stop communication and everything by purpose. That's what kill() is about.
Because kill() was watered-down we added minkill() to get the pure thing back.  
Don't give me reason to add something like "kill_kernel()".

AnHardt, 2019-02-19:
Questionable is, if kill(() is always the right thing to do, when we call it now.
In `validate_homing_move()` for sure a kind of "stop_all_motion_until_M999" is enough.

marcio-ao, 2019-02-19:
@AnHardt: It looks like the issue I was seeing was caused by our host program filling up the output buffer, which causing it to block on write and stop reading messages from Marlin. My original assumption that Marlin was not sending messages was incorrect and the additional calls to `HAL_idletask()` did not really make any difference. I am closing this ticket.

closed 2019-02-19 by marcio-ao