MarlinFirmware/Marlin #13209: Z offset adjust - err too far

HuguesDug, 2019-02-19:
Config:
Marlin 1.1.9 bug fix (was previously 1.1.9, 1.1.8 and manufacturer firmware).
Anet A8 with bed leveling sensor (manufacturer)

Problem:
I cannot adjust z offset from the LCD screen.
It returns a message: err too far.

No clue how to fix that.

As a workaround, I use Arduino IDE, change the value Z_PROBE_OFFSET_FROM_EXTRUDER in firmware, upload, M502 and M500.
It works, but is everything but easy and fast.

Any solution so the LCD menu do not return an error message ?

boelle, 2019-02-20:
Please test with the latest `bugfix-2.0.x`

HuguesDug, 2019-02-20:
I will... but I am not sure It will fit inside the EPPROM of the Anet board ! I am already full at 99% with current options.

boelle, 2019-02-20:
then pick important options first or replace with better board :-D

thinkyhead, 2019-02-20:
If you see that error, you're not adjusting the Z probe offset value. You're adjusting the `M206` (aka `M428`) "Home Offset" value.

closed 2019-02-20 by thinkyhead

HuguesDug, 2019-02-20:
Hello thinkyhead.

My printer is in French :-)  So I guess your translation is the right one "home offset value". 
#define MSG_SET_HOME_OFFSETS                _UxGT("Regl. decal. origine")

And yes, it seams it is related to "M206". So, what is the cause of the error, and how to fix it ?

thinkyhead, 2019-02-20:
You most likely don't really want to use home offsets, so just don't use them. If you actually must use them, you have to be within 2cm of an endstop to use that menu item or `M428`.