MarlinFirmware/Marlin #13210: ramps shield 1.4/ arduino marlin 1.1.9 hotend or heatbed  will not try to heat up

patmckenz, 2019-02-19:
Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

ramps 1.4/ arduino marlin 1.1.9 hotend or heatbed  will not try to heat up
### Steps to Reproduce

1. was upgrading from marlin 1.1.7
2. try marlin 1.1.9 and bugfix all seems to work but hotend and heated bed 
3. try to heat up hot end and bed nothing happen then get heater shutdown error.
reloaded marlin 1.1.7 all works.

**Expected behavior:** [What you expect to happen]

[Marlin-1.1.x2-7-19.zip](https://github.com/MarlinFirmware/Marlin/files/2880138/Marlin-1.1.x2-7-19.zip)

boelle, 2019-02-20:
Please test with the latest `bugfix-2.0.x`

thinkyhead, 2019-02-25:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

closed 2019-02-25 by thinkyhead