MarlinFirmware/Marlin #13214: READ THIS BEFORE YOU POST

boelle, 2019-02-20:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.
Before posting a bug report please test with `bugfix-2.0.x` to check if problem is gone

boelle, 2019-02-20:
@thinkyhead please pin this post and remove this

boelle, 2019-02-20:
@thinkyhead and if its possible also block comments

reloxx13, 2019-02-20:
this is not a forum, u cannot block comments or pin an issue.

this belongs in the issue_template file inside of the .github folder and those infos are already there, as u should have seen them urself by hitting the "new issue" button and writing this textwall.

boelle, 2019-02-20:
then why do we have a pinned issue?

boelle, 2019-02-20:
[image]

boelle, 2019-02-20:
problem is that people dont read that template text

on the other hand i know that octoprint github have something that checks if people are filling out the form correctly and rejects the issue if its not done correct

and the real problem is that we have an issue list 900+ long where a lot of the things do not belong here

reloxx13, 2019-02-20:
yep, a maintainer has to strictly close those issues to minimize it on bugs.

we had the same problem in the tasmota github. if issues get closed fast for being a question, ppls will learn to post at the right places.

Roxy-3D, 2019-02-20:
I prefer a different tone...   

Support requests will usually get a quicker and more complete response over on the Discord channel.

But I'll help people where ever I see the question pop up.   And the same goes for most of the people that contribute here.

reloxx13, 2019-02-20:
u can still answer/help them. 

set a "question" tag and close.
if u want to help ppls u move to closed issues and look/filter for the "questions" tag.

boelle, 2019-02-20:
i agree... if a question should find its way to here it can still be answered and used even if closed

the point is to keep the list of open issues short with bugs so these can be fixed

i know @Roxy-3D just filters the list so the last updated goes to the top, it does not solve the problem thou where the list is filled with things that are solved long time ago, etc etc etc

psavva, 2019-02-21:
We really should go back to using labels, that used to work just fine ..

boelle, 2019-02-21:
it also helps if people post questions elsewhere if they can
but also a lot of the open issues have been solved but people dont close them

it makes the issue list a real big mess

i have given @thinkyhead a list of issues that does not have labels, i have worked myself from last page up to page 30, but i will give him a bigger list when he asks for it

closed 2019-02-22 by boelle

AnHardt, 2019-02-22:
Uhhh. This issue is open since 2 days. It was wot commented the last 18h.

Is it already fixed?
Try the latest bugfix branches.
Please close.

Roxy-3D, 2019-02-22:
@AnHardt has a good sense of humor.

thinkyhead, 2019-02-22:
The OP is my customary boilerplate reply posted to issues that can be addressed by peers in the community, and the aim is to help us to keep the issue queue limited as much as possible to issues more directly related to Marlin development. It's not meant to be a masthead, because some questions point to legitimate issues or documentation holes, so we just ask users to apply their discretion. If something seems to be more suited to community support, or we are very busy, we might hand you off to the community experts. But sometimes if the issue is simple and we can answer with a link, then we don't mind taking small amounts of time. We can't always fix your specific issue "now" because we works sporadically and have long todo lists. But the more we can pass off direct support the quicker we can get to all pending tasks, with bug-fixes at the top of the list.