MarlinFirmware/Marlin #13215: JUNCTION_DEVIATION can't be enabled in Deltas

jmdearras, 2019-02-20:
### Description

if you enable JUNCTION_DEVIATION in a Delta config, the compile fails in configuration_store at saving X, Y, Z jerk values as they do not exist.  This is because conditionals_post in line 46 defines HAS_CLASSIC_JERK as IS_KINEMATIC is true for Deltas.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2884913/Configuration.zip)

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

reloxx13, 2019-02-20:
try again with latest 2.0.x source.

got fixed today

https://github.com/MarlinFirmware/Marlin/issues/13000

jmdearras, 2019-02-20:
It works. thanks!

closed 2019-02-20 by jmdearras