MarlinFirmware/Marlin #13220: WRITE requires 2 arguments, but only 1 given: hal_timers_stm32f7

dekra54, 2019-02-21:
I'm currently trying to implement the nucleo H743 as a new Board and stumbled across this line in the HAL for the STM32F7  which i used as a starting point. I dont know if this is already an issue in the F7 version or it only pops up because i'm trying to use the stm32duino framework instead of the rogerclark one.

https://github.com/MarlinFirmware/Marlin/blob/9f5e4e5ad812e39f846759283c1f8de2c475f7d2/Marlin/src/HAL/HAL_STM32F7/HAL_timers_STM32F7.cpp#L89

AnHardt, 2019-02-21:
In this change
```
      HAL_NVIC_SetPriority(timerConfig[0].IRQ_Id, 1, 0);
      pinMode(STEPPER_ENABLE_PIN,OUTPUT);
      digitalWrite(STEPPER_ENABLE_PIN,LOW);
      SET_OUTPUT(STEPPER_ENABLE_PIN);
      WRITE(STEPPER_ENABLE_PIN);
      break;
```
the LOW got lost.

dekra54, 2019-02-21:
Thank you
For some reason i could not see the history for this file and thereby this change was not visible.
Did this happen for a reason eg. some other part of code has changed or just a "typo"?

AnHardt, 2019-02-21:
Typo i think.
----
Go to https://github.com/MarlinFirmware/Marlin/blob/9f5e4e5ad812e39f846759283c1f8de2c475f7d2/Marlin/src/HAL/HAL_STM32F7/HAL_timers_STM32F7.cpp#L89
Klick [Blame} at the top.
Go back to line 98.
The [link at the left](https://github.com/MarlinFirmware/Marlin/commit/55807731918625b527a75668bc7fa069640f83b9) will send you to the commit this was last changed.
With some luck you'll find the reason for that commit. 
If the actual change does not fit to the reason its likely to be a typo, especially when making ''no'' sense.

dekra54, 2019-02-21:
Thanks for the detailed explanation. I will look further into it and close this issue for now. If needed i edit it and add a pull request.

closed 2019-02-21 by dekra54