MarlinFirmware/Marlin #13222: MKS Gen V1.4 pin assignments

michaelj1965, 2019-02-21:
I am trying to do the Arduino example of an analog input controlling a motor.

Stepper(steps, pin1, pin2)
Stepper(steps, pin1, pin2, pin3, pin4)
Description
This function creates a new instance of the Stepper class that represents a particular stepper motor attached to your Arduino board. Use it at the top of your sketch, above setup() and loop(). The number of parameters depends on how you've wired your motor - either using two or four pins of the Arduino board.

But, I cannot seem to find the proper pin numbers and there is scarce information on the aforementioned board.

 Stepper(steps, pin1, pin2, pin3, pin4)
I get some pulsing if I use Stepper (100, 46, 48, 62, 40) but no real movement.

AnHardt, 2019-02-21:
This question has low relationship with Marlin.

You can find Marlins pin assignments in the pinfiles.
That will not help you. Marlins stepper drivers are working with enable, step and pulse signals.
The Arduino library and examples expect 4 pins driving the transistors of the bridge directly.
That's completely incompatible.

michaelj1965, 2019-02-21:
I have Marlin loaded on the machine and it is working. That is the only place that references this particular board. Therefore it seems to be a Marlin question. If there is another website that can answer the question then please direct me. I have researched the Marlin pin files and cannot get the answer I need.

AnHardt, 2019-02-21:
https://www.arduino.cc/en/reference/stepper and follow all links. Have a cloth look on the  ICs they are using.
Than have a look into any datasheet for stepper drivers used by Marlin.

AnHardt, 2019-02-21:
Compare [L297](https://www.st.com/resource/en/datasheet/l297.pdf) and [L298](https://www.mouser.de/datasheet/2/389/l298-954744.pdf).
L297 is connectable to Marlin and that provides the signals for the L298.
Arduino Library provides the signals for the L298 directly,  but the L297 is much better in that. 

This is the Marlin bug tracker, for pull requests and for discussing code.
For general Arduino, technical or beginners questions please go elsewhere.
Please close this issue.

michaelj1965, 2019-02-21:
da hilft mir nich...

AnHardt, 2019-02-21:
Natürlich nicht. 
Weil schon der Ansatz Blödsinn ist. Das ist alles was ich hier erklären kann. Für einen Grundkurs in Schrittmotortreibern habe ich nicht die Muße. Deswegen müssen Dir die paar Tips für die weitere Recherche reichen.
Nochmal im Klartext:
Mit der Arduino Libray kannst Du nicht die gleichen Schrittmotor Treiber ansteuern wie mit Marlin. Die Suche nach 'passenden' Pins ist hoffnungslos.

AnHardt, 2019-02-21:
@thinkyhead 
Would you please be so kind to close this? The user deleted himself and will not be able do do that.

boelle, 2019-02-22:
agreed, strange way to behave

closed 2019-02-25 by thinkyhead