MarlinFirmware/Marlin #13224: Display print time HH:MM:SS possible in 1.1.x?

n9jcv, 2019-02-22:
Has there ever been a request to have the current print time display the seconds as well as hours and minutes?  I looked thru the configuration but did not see any options like this.

Thanks
Bruce

AnHardt, 2019-02-22:
Yes.
But what for?

boelle, 2019-02-22:
and go for bugfix 2.0, its here things happens these days

thinkyhead, 2019-02-22:
Marlin is open source so that customizations like this can be made for users that really want them. But this is not something we will be rolling into the main Marlin as a new feature. You should ask around the - RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415) for help if you're not sure how to make the needed modifications to get seconds to show on the display.

closed 2019-02-22 by thinkyhead