MarlinFirmware/Marlin #13225: Missing labels?

boelle, 2019-02-22:
since i dont have the powers to add labels myself who other than @thinkyhead and @Roxy-3D can do this?

does it even matter?

TheSFReader, 2019-02-22:
Almost no relation to the subject, but I saw some "bad" comments on your work to ensure the "Issues" list is cleaned up, and I think you're doing a good and needed work there. Thanks :-)

boelle, 2019-02-22:
well its a bit related, and i just tried to help where i can

my tone might sound rough but there have been about 150 open issues that is closed

- some where fixed but not closed
- some of them people have moved on to other firmware and because of that they are not able to test 
  and verify, is it reasonble that @thinkyhead can test all hardware etc? i think not
- many of them was questions that belong to either discord or facebook

and i made a few mistakes, but i have figured i have read about 1000 issues at the end my head is feeling like a sponge and then you screw up a bit

thinkyhead, 2019-02-22:
I would prefer fewer labels, as I don't want to start having a label for every possible topic. There have been some very active topics like SD printing and so on, so those merited labels for posterity, and they still come into use.

What labels do you think are missing?

boelle, 2019-02-22:
oh, bad wording

it was issues that dont have any labels at all, i posted a rough list of pages that are missing labels on discord as a DM

boelle, 2019-02-22:
like the first pages dont have issues with labels on, but the last pages nearly every issue has been given a label

closed 2019-02-23 by thinkyhead