MarlinFirmware/Marlin #13230: [2.0.x] SD Disabled, Pause SD Print is showing

reloxx13, 2019-02-22:
### Description

I disabled SD Support in the configuration and still "Pause SD Print" (SD Druck pausieren) is showing up in the menu while printing.

### Steps to Reproduce

- Start a print with octoprint (over USB)
- Hit the knob to enter the configuration
- Pause SD Print is showing up

**Expected behavior:**
Nothing SD related should show up if SD Support is disabled.
Especial in the menus due strings are eating too much flash size.

**Actual behavior:**
SD Pause Print menu entry showing up

#### Additional Information

Im on 2.0.x-bugfix from Feb 15, 2019.

Printer: Ender 3 with orginal Board. TL Smoothers on X and Y. ABL added.

You can find my fw configuration here on GitHub:

https://github.com/reloxx13/Marlin/blob/my-marlin/Marlin/Configuration.h
https://github.com/reloxx13/Marlin/blob/my-marlin/Marlin/Configuration_adv.h

thinkyhead, 2019-02-23:
Marlin now has the ability to pause the host.

closed 2019-02-23 by thinkyhead

reloxx13, 2019-02-23:
what about renaming the entry. if u can detect that its printing from usb, remove the "SD" string. 

low prio with this.

ill try to pause a test print again, did not work last time.

br

thinkyhead, 2019-02-25:
Indeed. The `SD` should be removed for this case — and maybe in-general.

reloxx13, 2019-02-25:
> Marlin now has the ability to pause the host.
but this requires HOST_ACTION_COMMANDS enabled, right? without it and without sd_support we dont need "pause sd print".

i tried it without host_action_commands and it shows the filament change pause menu, but the (octoprint) print did not pause (cuz i dont have host actions configured).

thinkyhead, 2019-03-03:
Let's see if #13293 helps.