MarlinFirmware/Marlin #13231: [2.0.x] Cooldown does not reset fan speed value

reloxx13, 2019-02-22:
### Description
if i change filament, i heat the nozzle with the lcd and also enable the parts cooling fan (fan speed 255) to avoid stringing when i remove oouzing filament from the nozzle.

After im done, i use "cooldown" from the menu. this also disables the fan.
But the LCD is still showing the fanspeed at 255.

### Steps to Reproduce

1. Heat nozzle
2. Set Fan to 255
3. Hit cooldown

**Expected behavior:** [What you expect to happen]
Fan Speed should go down/show 0 again.

**Actual behavior:** [What actually happens]
Fan speed stays at 255

#### Additional Information
Im on 2.0.x-bugfix from Feb 15, 2019.

Printer: Ender 3 with orginal Board. TL Smoothers on X and Y. ABL added.

You can find my fw configuration here on GitHub:

https://github.com/reloxx13/Marlin/blob/my-marlin/Marlin/Configuration.h
https://github.com/reloxx13/Marlin/blob/my-marlin/Marlin/Configuration_adv.h

reloxx13, 2019-02-22:
Nice2Have:
Would be also nice if the fan speed stays on/get enabled till "min extrusion temp" is reached when using cooldown to have a faster cooldown with less ouzing.

thinkyhead, 2019-02-23:
The extruder auto fan will stay on above a certain temperature. That's the best option to use for your hotend heatsink fan. The print cooling fan only needs to be on during active printing, and `M107` at the end of the print can ensure that the print fan is turned off under normal circumstances.

I've applied a patch to `bugfix-2.0.x` just now which _should_ actually turn off the print cooling fan(s) when you click the "Cooldown" menu item. Please let me know if it works for you.

AnHardt, 2019-02-23:
Switching fans off for cooling is counter-intuitive.

reloxx13, 2019-02-23:
@thinkyhead i think u didnt understand me correct.

it would turn off the fan if i hit cooldown, but it did not set the value to 0 on the lcd.

@AnHardt do u want to cooldown the bed with the partsfan? 

i think it should be 100% fanspeed on cooling down the hotend/nozzle.

--

ill flash tomorrow, printing now :p

thinkyhead, 2019-02-25:
> it would turn off the fan if i hit cooldown, but it did not set the value to 0 on the lcd.

And is that now fixed?

> i think it should be 100% fanspeed on cooling down the hotend/nozzle.

The print cooling fan points away from the nozzle, so there is no point in leaving it on to help cool the hotend. And it is not reliably associated with the bed, either. It seems better to just turn off print-cooling in the cases of heating abort, since it is tightly associated with cooling the object during active printing.

reloxx13, 2019-02-25:
> > it would turn off the fan if i hit cooldown, but it did not set the value to 0 on the lcd.
> 
> And is that now fixed?

YES! 👍 
 
> > i think it should be 100% fanspeed on cooling down the hotend/nozzle.
> 
> The print cooling fan points away from the nozzle, so there is no point in leaving it on to help cool the hotend. And it is not reliably associated with the bed, either. It seems better to just turn off print-cooling in the cases of heating abort, since it is tightly associated with cooling the extruded filament during active printing.

guess this could end in a long discussion :D
the airflow under the nozzle will also take away the heat from the nozzle and at the same time reduce oouzing filament.
as said, this is nice2have, if u wont add it, close this urself pls :)
ot ping me here and ill create a new clean FR issue with it.

closed 2019-03-05 by thinkyhead