MarlinFirmware/Marlin #13233: M852 doesn't change anything

Entretoize, 2019-02-22:
### Description

I'm adding the M852 code in my gcode files like that:

`M852 0 0 0.9 0`

And it changes nothing, I printed a test cube that is exactly the same with or without the code.
Of course I modified the firmware with that:

```
#define SKEW_CORRECTION

#if ENABLED(SKEW_CORRECTION)
  // Input all length measurements here:
  #define XY_DIAG_AC 282.8427124746
  #define XY_DIAG_BD 282.8427124746
  #define XY_SIDE_AD 200

  // Or, set the default skew factors directly here
  // to override the above measurements:
  #define XY_SKEW_FACTOR 0.0

  #define SKEW_CORRECTION_FOR_Z
  #if ENABLED(SKEW_CORRECTION_FOR_Z)
    #define XZ_DIAG_AC 282.8427124746
    #define XZ_DIAG_BD 282.8427124746
    #define YZ_DIAG_AC 282.8427124746
    #define YZ_DIAG_BD 282.8427124746
    #define YZ_SIDE_AD 200
    #define XZ_SKEW_FACTOR 0.0
    #define YZ_SKEW_FACTOR 0.0
  #endif

  // Enable this option for M852 to set skew at runtime
  #define SKEW_CORRECTION_GCODE
#endif
```

### Steps to Reproduce

1. Add M852 code to your gcode
2. Print

Ludy87, 2019-02-22:
Syntax: `M852 [I] [J] [K] [S]`
See http://marlinfw.org/docs/gcode/M852.html

closed 2019-02-22 by thinkyhead

Entretoize, 2019-02-23:
Can you reopen ? I tested with M852 I0 J0 K0.9 S0 and this is the samedi.

Entretoize, 2019-02-24:
I was wrong, sorry , it works