MarlinFirmware/Marlin #13238: BL touch not working

Telekmac, 2019-02-23:
I was just using my bl touch and I have had the configuration set and have e=been printing successfully with it for months. But just recently the probe stopped deploying all together. The self test fails and anything that requires deploying it fails. If i open it myself and do the reset command it will retract but not deploy. No firmware changes have been made besides readjusting the z offset.

psavva, 2019-02-23:
Hi Telekmac, 

So you're saying you had no firmware changes.
Your probe is failing to deploy.  Sounds like a BLTouch hardware issue.

Start by checking all the wiring and also check the PIN.  Why did you need to adjust the z-offset after months of successful printing...  Maybe the    is loose/dropped...

boelle, 2019-02-23:
what @psavva said ^^^

mr-miky, 2019-02-24:
It could also be a PWM problem on the control pin

boelle, 2019-02-24:
could a to cheap board or controller board with badly picked parts be part of the problem? or maybe a noisy psu?

einencool, 2019-02-25:
I had the problem, with a cheaper 3D Touch that the wire to the solenoid was broken. It was broken under the original tape at the top of the device...

It sounds a little bit similar..

Telekmac, 2019-02-25:
I will be checking the wires for contunity tomorrow. I had to adjust the
offset fue to installing a glass bed. I dont think it is a board issue but
i will check.

closed 2019-03-03 by thinkyhead