MarlinFirmware/Marlin #13244: Compiile error UBL together with SERIAL_PORT -1

dystechnic, 2019-02-23:
I use a DUE based controller (Ultratronics) and when I configure #define SERIAL_PORT -1 together with #define AUTO_BED_LEVELING_UBL I run into a compile error. Since it is a different error than #7669 I opened a new bug report.

----->
In file included from Marlin/src/feature/bedlevel/ubl/../../../inc/MarlinConfig.h:49:0,
from Marlin/src/feature/bedlevel/ubl/ubl.cpp:23:
Marlin/src/feature/bedlevel/ubl/ubl.cpp: In static member function 'static void unified_bed_leveling::display_map(int)':
Marlin/src/feature/bedlevel/ubl/../../../inc/../core/serial.h:126:55: error: 'class MarlinSerialUSB' has no member named 'flushTX'
#define SERIAL_FLUSHTX()                MYSERIAL0.flushTX()
^
Marlin/src/feature/bedlevel/ubl/ubl.cpp:253:9: note: in expansion of macro 'SERIAL_FLUSHTX'
SERIAL_FLUSHTX();
^
*** [.pioenvs/DUE_USB/src/src/feature/bedlevel/ubl/ubl.cpp.o] Error 1
<-----

I tried it with the latest version (23-02-109).
If I change either of the mentioned options the compile error is gone. 
So if I change the serial port to 0 **or** change UBL to Mesh for example then it compiles succesfully.

Attached are my config files.
[configuration.h](https://ghostbin.com/paste/oa4zf)
[configuration_adv.h](https://ghostbin.com/paste/4dj7d)

thinkyhead, 2019-02-25:
`MarlinSerialUSB` has no `flushTX` method. So you must set `TX_BUFFER_SIZE` to 0.

dystechnic, 2019-02-25:
That did the trick.
Thank you!

closed 2019-02-25 by dystechnic