MarlinFirmware/Marlin #13245: ??bug?? inconsistent movement

ruggb, 2019-02-23:
I don't know if this is a bug or I have something set wrong.
my Z endstop is a nozzle activated piezo with an M48 std dev of 0.0229.
I have set my Z offset to 0.05 in the LCD menu. When I home Z, Repetier shows Z = -0.05. When I step Z by 0.01, Repetier shows it going from -0.05 to 0.00. The 2nd step goes to 0.01 etc. If I step Z by 0.1 initially, Repetier shows Z = 0.05;  a 1.0 step shows 0.95 in Repetier.
This seems like an inconsistency with Marlin. I would expect it to either take 5x 0.01 steps to get to 0.00 based on how it handled the 0.1 and 1.0 steps, or any single initial step to add the Z offset and go to that position. IE, a 0.01 step goes to 0.01, a 0.1 step goes to 0.1, a 1.0 step goes to 1.0, etc.
A 0.08 shim test appears to yeild the same result for home Z, step 9x 0.01 or home Z, step 1x 0.1 + 3x 0.01. Both cases show 0.08 in Repetier.
What Repetier shows appears to be correct, but obviously 0.01 is difficult to assess with a shim.
How is Marlin "suppose to" handle a Z offset?

reloxx13, 2019-02-23:
rly hard to read, would be nice if u format it better for the devs.

boelle, 2019-02-23:
also attach 2 config files as zip

boelle, 2019-02-23:
and make sure its not a hardware problem

ruggb, 2019-02-24:
Is this any better?
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2897487/configs.zip)

;
using v2 bugfix [f6ab62b](https://github.com/MarlinFirmware/Marlin/commit/f6ab62bc13ddca007b031047b55d2d67ad40ca4e)
;
coreXY printer has a Z endstop/probe nozzle activated by a piezo with an M48 std dev of 0.0229.
;
I have set my Z offset to 0.05 in the LCD menu. 
;
When I home Z, Repetier shows Z = -0.05. I think this is correct because of the offset.
I step Z by 0.01, Repetier shows Z going from -0.05 to 0.00.
The 2nd 0.01 Z step goes to 0.01, then the next 0.01 step => 0.02, etc. 
;
If I step Z by 0.1 initially (after a home), Repetier shows Z = 0.05; an initial 1.0 step Repetier shows Z = 0.95 
;
This seems like an inconsistency with Marlin. I expected it to either take 5x 0.01 steps to get to 0.00 based on how it handled the 0.1 and 1.0 steps.
;
I actually thought it would eliminate the offset on the first step by adding the Z offset and going to that position. IE, with a 0.05 offset, a 0.01 step goes to 0.01 (actual step = 0.06), a 0.1 step goes to 0.1 (actual step = 0.15), a 1.0 step goes to 1.0 (actual step = 1.05)
;
Using a 0.08 shim to test, it appears to yeild the same result if I home Z, step nine 0.01 or if I home Z, step once 0.1 + three 0.01. Both cases show 0.08 in Repetier. Why does it only get to 0.00 on one 0.01 step, but gets to 0.05 on one 0.1 step and 0.95 on one 1.0 step?
;
What Repetier shows appears to be correct based on the shim, but obviously 0.01 is difficult to assess with a paper shim.
;
How is Marlin "suppose to" handle a Z offset?

AnHardt, 2019-02-24:
If "software endstops" are on, they are established with homing.
If beyond the endstops the next move will bring you to the border.
- Could be an explanation.

thinkyhead, 2019-02-24:
`MIN_STEPS_PER_SEGMENT`…?

thinkyhead, 2019-02-24:
If the host is showing the position changed to 0, then I assume it gets it from `M114`… So Marlin is constraining Z, but it is not reporting this to the host at the earlier time, or the host is not updating its display. Connect with the dumb serial terminal in Arduino and see what `M114` reports at various times in the process.

ruggb, 2019-02-25:
I don't have a dumb terminal. M114 is showing the same info via Repetier so Repetier must be getting it from that. Otherwise, it would probably report -0.04 after the 1st 0.01 step.

When I home Z, Marlin thinks it is at 0.00, except that I told Marlin it is really at -0.05. Why would Marlin constrain anything except to keep it from going more negative.

based on the paper test, Repetier appears to be reporting correct data.
If the first 0.01 step goes to 0.00, would one not expect any other 1st step increment to do the same thing == go to 0.00?
If the other step increments go to Zoffset + step increment, shouldn't one expect the same thing for a 0.01step?
Wouldn't it be better if a step from "home" went to the step value? If "home" is 0.00 then it does. If there is a Zoffset it does not for any step. Since the paper test is consistent, it appears that Marlin is not.

Does this happen with your printers? Are any CoreXY?

thinkyhead, 2019-02-25:
Ok, thanks for the info. We'll continue to look into it.

AnHardt, 2019-02-25:
If you are using Repetier Host you have a 'dumb enough' terminal. Just switch to the tab with the serial log and adjust to put out everything.

ruggb, 2019-02-25:
So there is a difference between a STEP and a GOTO when dealing with Zhome+offset.

So who is in control, because I don't think there should be a difference, do you?

Repetier knows it is at -0.05 after a home
Marlin knows it is at -0.05 after a home
Clicking a STEP in Repetier gives inconsistent results when Z=home with an offset (Z=-0.05):
a 0.01 step yields a step of 0.05 to Z=0.00 < this is especially strange, no logic here
a 0.1 step yields a step of 0.1 to Z=0.05 < this I understand, but from Z=home+offset it should go to 0.1
a 1.0 step yields a step of 1.00 to Z=0.95 < as above but should go to 1.0
a command of G1 Z0.01 yields a step of 0.06 to Z=0.01
a command of G1 Z0.1 yields a step of 0.15 to Z=0.1
a command of G1 Z1 yields a step of 1.05 to Z=1.00

????

AnHardt, 2019-02-25:
Normally Marlin is in absolute mode. The last 3 G1 commands go to a absolute position. Nothing strange.
Repetier moves in relative mode. If you'd have a look into the log , as you where told to, you'd see the `G91` before and `G90` behind every (Repetier button klick) move.
For the two larger moves everything is as expected.
The first small 0.01mm move would not reach 0.0. If the software endsop is `on` and at 0.0, what i suppose it to be, a target coordinate below the endstop is not allowed and automatically corrected to be on the border, on 0.0.

ruggb, 2019-02-25:
@AnHardt Where do you think I got the info I just posted? - I LOOKED INTO THE LOG.
I would appreciate less condescending.
Repetier is not showing any G91 or G90 commands.
for the 1st 0.01 step after home it shows N## G1 Z0 F1000*63
for the next one it shows N## G1 Z0.01 F1000*71
Why does it want to step to Z0 for the first Z0.01 step?
Why does it want to step to Z=0.05 for the first 0.01 step?
Even if it were issuing G90/91 commands the commands I see do not make sense.

AnHardt, 2019-02-25:
If the sent command is already `G1 Z0 F1000` it's not Marlin but Repetier.
Have a look on Repetiers `Printer Settings`. Repetier has its own software endstops.

ruggb, 2019-02-25:
if RH is in relative mode a G1 Z0 would cause no movement.
Since it moves to Z0 for a 0.01 step command it is in absolute mode.
Since it knows it is at -0.05 it SHOULD treat the first step command after a home as a step from Zoffset to Z0 + step.
The problem is, it appears to be INCORRECTLY trying to take Zoffset into account but only for a 0.01 step. It needs to do that CORRECTLY for any step after home. For the 0.1 and 1.0 step it appears to be in relative mode. BUT since no G90/91 commands are issued it is doing the math itself.

ruggb, 2019-02-25:
@AnHardt Well that is what I said a few posts back. It is a Repetier issue.
RH does not have any setting for Zoffset. The HOME settings are all 0 If I change Z home to -0.05 it makes no difference. The first 0.01 step command is still G1 Z0

ruggb, 2019-02-25:
There is nothing you can do. I have posted the issue on the RH forum. Is there a better place to get RH dev response?

AnHardt, 2019-02-25:
How about setting Repeties to
[image]

ruggb, 2019-02-25:
yes, that is the setting I was talking about and nothing I do to it changes anything. The RH dev said it is a problem and put it on his to do list. But I am not sure what he said he was to do will fix it. 
Somehow he is associating the problem as being a fixed Z min setting. But based on what it does for 0.01 step vs 0.1 and 1 I think something else is wrong. He needs to read the Z value after homing and use that in calc the 1st step. Maybe that is what he intends on doing. I am a HW engineer and sometime you devs don't make any sense to me.

AnHardt, 2019-02-25:
If you are now sure it's a Repetier issue - please close this one.
Providing a (commented?) serial log would have seed up this a lot.

ruggb, 2019-03-02:
Yes, I thought I did close it.
thanks

closed 2019-03-02 by ruggb