MarlinFirmware/Marlin #13248: [FR] Silent mode

boelle, 2019-02-24:
could be cool if marlin could support "silent" mode like prusa does

from my  understanding all they do is having 2 set of movement settings and you can select them from either display or from their slicer

i found where the different values are in their config: https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h#L90

Not sure how to better descripe it, but i know that it has not been inplemented yet

Vertabreak, 2019-03-10:
i changed movement speeds to be quieter in my configuration ill wait the 3 seconds extra to not hear the screeching.

boelle, 2019-03-10:
Nice to hear that we can go a bit more silent just by changing movement speed

Vertabreak, 2019-03-11:
The noise reduction adjusting xy homing to 2k and z to 900 is very noticeable. You can verify this yourself if you like. I use the settings on all 5 of my printers. Adaptive smoothing feature also seem to help.

boelle, 2019-03-11:
it could be nice if we like on prusa can change between 2 settings while printing etc.

but i guess we can change settings in eeprom until someone has figured how to hammer up a PR for it

Vertabreak, 2019-03-11:
Can easy set up 2 custom commands to set XYZ feed rates one high one low with m203.

boelle, 2020-04-25:
@thinkyhead is  this idea worth the trouble?

thinkyhead, 2020-04-25:
Sure! It shouldn't take very long at all. It's just adding one or two simple menu items, right? Basically a cut-and-paste job. I'm dealing with too many things to do it myself, but put out the call if you need help.

boelle, 2020-04-25:
Correct and maybe an MCode to switch

closed 2020-06-21 by boelle