MarlinFirmware/Marlin #13249: [FR] Support for Slic3r PE M73

boelle, 2019-02-24:
as the title says we should support slic3r PE's version of M73 which has the following format

[image]

**M73 Pxxx Qxxx Rxxx Sxxx**
where P is the percentage of the total time already printed in normal mode, R is the time remaining for normal mode in minutes, Q is the percentage of the total time printed in the stealth mode, and S is the time remaining to the end of print in the stealth mode.

brandstaetter, 2019-02-24:
Does Slic3r PE always write it out like that or only if the silent mode support is active?
[image]

boelle, 2019-02-24:
in this one  https://github.com/MarlinFirmware/Marlin/issues/8515#issuecomment-466764196  @einencool wrote:

> My printer is busy for the next hour, so I can't check it, but this was found in the internet
> 
> **M73 Pxxx Qxxx Rxxx Sxxx**
> where P is the percentage of the total time already printed in normal mode, R is the time remaining for normal mode in minutes, Q is the percentage of the total time printed in the stealth mode, and S is the time remaining to the end of print in the stealth mode.
> 
> And in my Code from Slic3r it shows The M73 Pxxx and Rxxx because I disabled the "Silent Mode"

boelle, 2019-02-24:
i have not tested it myself but i can do so tomorrow, ie generate some sample Gcode and post it here as a zip

boelle, 2019-02-24:
btw... i run my printer as if it was an mk2.5, firmware has same movement settings etc as the mk2.5

the only changes i have is start and stop gcode and i print with 3 walls and not 2. other that its stock slicer settings

revilor, 2019-02-24:
When you enable "Supports remaining times" in Printer Settings the gcode will contain M73 P.. R... When you enable "Supports silent mode" it will add additional M73 Q.. S...

boelle, 2019-02-24:
more or less the same @einencool said :-D so i was not far off

einencool, 2019-02-24:
Oh, I think there is a little mistake ;-)

In the picture in post number 1 there is the normal status bar and the text at the end was only a 
M117 command 
which was a replacement for the original M73 Pxxx Rxxx
I replaced it with Notepad++ and hoped, that it will count down in the status bar. 
The counter (ie 03:00) was there normal upcounting one...

But the M117 commands weren’t recognized any more, so it shows me all the time 185 min left...

Does anybody know how to get the variable for the „remaining print time“ or the complete print time counter??
Then I would just add this in the start gcode and will have the normal counter and in the status bar the whole time

boelle, 2019-02-24:
that is what we should figure out...

i wrote that 03:00 could be used as the remaing time in whatever mode the printer is in
i say whatever mode because we dont have silent mode supported, but it could be that we get it, at least we should account that we will have silent mode some day

boelle, 2019-02-24:
progress bar above 03:00 should jsut be updated with the % number from M73

boelle, 2019-02-24:
but city of rome was not build in a day so it will take some time i'm sure

but since i'm not a coder it will have to be done on the mercy of others

einencool, 2019-02-24:
The gcode from the picture doesn’t contain a M73 command. 
And in the config adv i haven’t activated the M73 option.
Yeah, would be great if it can work. But in my mind it would be even better to see the processing time AND the remaining time or the complete time.

In the gcode from slic3r PE is at the end the complete processing time , and when it would be possible to get this info on the screen instead of the program name, that would be great.
For me the Program name is not so important, Lately when the job is done, I know what it should be.....

boelle, 2019-02-24:
@einencool we are not quite talking about your issue even thou its close related...

but yes... both printed time and time remaing time would be optimal... and it would have to work on both text displays and graphics ones

boelle, 2019-02-24:
let's see what the code gods come up with

ilachEU, 2019-03-25:
basic implementation for DOGM (changed files only)
[M73.zip](https://github.com/MarlinFirmware/Marlin/files/3002744/M73.zip)

GMagician, 2019-10-17:
This may be closed now...I think

rmoravcik, 2019-11-16:
Remaining time support was merged to bugfix-2.0.x (https://github.com/MarlinFirmware/Marlin/commit/e110f5abce90d139a768242780780832b09aa297)

closed 2020-04-25 by boelle