MarlinFirmware/Marlin #13250: [BUG] Bugfix 2.0 e15354e387cc825390fc4eaaf3ae4b784a8fd588 does not compile

Bergerac56, 2019-02-24:
### Description

As the title says: bugfix 2.0 e15354e387cc825390fc4eaaf3ae4b784a8fd588 does not compile for me.

The problem seems to have been introduced by the changes made in marlin/src/core/serial.h. 
I did not have compiling problems with commit: 562f2321d282eb16d538e6098b0155e0d91e7e4a

Message received at end of compile:
In file included from Marlin\src\HAL\HAL_LPC1768\../../sd/../inc/MarlinConfig.h:49:0,
from Marlin\src\HAL\HAL_LPC1768\../../sd/cardreader.h:24,

from Marlin\src\HAL\HAL_LPC1768\main.cpp:17:
c:\marlin-bergerac56\marlin\marlin\src\core\serial.h:55:79: error: pasting "(" and ""\n\necho:%s (%dMhz) Initialized\n"" does not give a val
id preprocessing token
if (!serial_port_index || serial_port_index == SERIAL_BOTH) MYSERIAL0.WHAT(##__VA_ARGS__); \
^
c:\marlin-bergerac56\marlin\marlin\src\core\serial.h:73:33: note: in expansion of macro 'SERIAL_OUT'
#define SERIAL_PRINTF(args...)  SERIAL_OUT(printf, args)
^~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\main.cpp:92:5: note: in expansion of macro 'SERIAL_PRINTF'
SERIAL_PRINTF("\n\necho:%s (%dMhz) Initialized\n", isLPC1769() ? "LPC1769" : "LPC1768", SystemCoreClock / 1000000);
^~~~~~~~~~~~~
c:\marlin-bergerac56\marlin\marlin\src\core\serial.h:56:43: error: pasting "(" and ""\n\necho:%s (%dMhz) Initialized\n"" does not give a val
id preprocessing token
if ( serial_port_index) MYSERIAL1.WHAT(##__VA_ARGS__); \
^
c:\marlin-bergerac56\marlin\marlin\src\core\serial.h:73:33: note: in expansion of macro 'SERIAL_OUT'
#define SERIAL_PRINTF(args...)  SERIAL_OUT(printf, args)
^~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\main.cpp:92:5: note: in expansion of macro 'SERIAL_PRINTF'
SERIAL_PRINTF("\n\necho:%s (%dMhz) Initialized\n", isLPC1769() ? "LPC1769" : "LPC1768", SystemCoreClock / 1000000); 

Config files:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2897923/Marlin.zip)

Moyster, 2019-02-24:
it compiles fine for me, so I compared your config files and mine and your SERIAL_PORTs config seems to be the root of it, if I edit my configs with :  
#define SERIAL_PORT -1  
#define SERIAL_PORT_2 0  

I get your build errors.

thinkyhead, 2019-02-24:
This is now patched. Thanks for the report!

closed 2019-02-24 by thinkyhead

Bergerac56, 2019-02-25:
Thanks Scott, this is solved for me