MarlinFirmware/Marlin #13252: odd _BED_SIZE & AUTO_BED_LEVELING

Hiroshi-OssanNo2, 2019-02-24:
### Description

Marlin 2.0 Bugfix

Odd _BED_SIZE & AUTO_BED_LEVELING
compile err
### Steps to Reproduce
#define X_BED_SIZE 211 
#define AUTO_BED_LEVELING_*****
PlatformIO Home 2.0.0·Core 3.6.4 autocompile

Odd _BED_SIZE 209 211 213  every time err
Even _BED_SIZE 210 208 212 no err

**Expected behavior:** [What you expect to happen]
no error
**Actual behavior:** [What actually happens]
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\I2cEeprom.cpp.o
In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_AVR\HAL.cpp:29:
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1139:5: error: static assertion failed: RIGHT_PROBE_BED_POSITION is outside the probe region.
static_assert(RIGHT_PROBE_BED_POSITION <= MAX_PROBE_X, "RIGHT_PROBE_BED_POSITION is outside the probe region.");

Ludy87, 2019-02-24:
Configfiles?

Hiroshi-OssanNo2, 2019-02-24:
<details>
<summary>Configuration_ANYCUBIC_CHIRON.h</summary>
<pre>
<code>
/**
 * ANYCUBIC CHIRON Marlin 2
 * edit Configuration.h bottom
 * #include "Configuration_ANYCUBIC_CHIRON.h"
 * 
 * referred https://github.com/ANYCUBIC-3D
 */

//
// Hardware 
//
#define EEPROM_SETTINGS
#undef MOTHERBOARD
#define MOTHERBOARD BOARD_TRIGORILLA_14
#undef DEFAULT_KEEPALIVE_INTERVAL
#define DEFAULT_KEEPALIVE_INTERVAL 3  // Number of seconds between "busy" messages. Set with M113.
#define SDSUPPORT

#undef DEFAULT_NOMINAL_FILAMENT_DIA
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
#define USE_XMAX_PLUG
//
    #undef X_BED_SIZE
    #define X_BED_SIZE 211
    #undef X_MAX_POS
    #define X_MAX_POS X_BED_SIZE
    #undef Y_BED_SIZE
    #define Y_BED_SIZE 216
    #undef Y_MAX_POS
    #define Y_MAX_POS Y_BED_SIZE

#undef TEMP_SENSOR_BED
#define TEMP_SENSOR_BED 1

// MIN_ENDSTOP
#undef X_MIN_ENDSTOP_INVERTING
#define X_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#undef Y_MIN_ENDSTOP_INVERTING
#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#undef Z_MIN_ENDSTOP_INVERTING
#define Z_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
// MAX_ENDSTOP
#undef X_MAX_ENDSTOP_INVERTING
#define X_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#undef Y_MAX_ENDSTOP_INVERTING
#define Y_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#undef Z_MAX_ENDSTOP_INVERTING
#define Z_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
// Z probe is blue switch
#undef Z_MIN_PROBE_ENDSTOP_INVERTING
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the probe.

// stepping motor
#undef INVERT_X_DIR
#define INVERT_X_DIR true // CHIRON
#undef INVERT_Y_DIR
#define INVERT_Y_DIR false // CHIRON
#undef INVERT_Z_DIR
#define INVERT_Z_DIR true // CHIRON

#undef INVERT_E0_DIR
#define INVERT_E0_DIR false // CHIRON
#undef INVERT_E1_DIR
#define INVERT_E1_DIR true // CHIRON Z2

#define DRVER_CHIP A4988 // trigorilla default
[... 159 lines cut ...]
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _XMAX_
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0
  #endif
#endif

#undef Z_HOME_BUMP_MM
#define Z_HOME_BUMP_MM 5
#undef HOMING_BUMP_DIVISOR
#define HOMING_BUMP_DIVISOR { 4, 4, 8 }  // Re-Bump Speed Divisor (Divides the Homing Feedrate)

//#define DEFAULT_MINSEGMENTTIME        20000 //chiron130
#undef DEFAULT_MINSEGMENTTIME
#define DEFAULT_MINSEGMENTTIME        50000

#undef DIGIPOT_MOTOR_CURRENT
#define DIGIPOT_MOTOR_CURRENT { 135,135,135,135,135 }   // Values 0-255 (RAMBO 135 = ~0.75A, 185 = ~1A)
//#define RX_BUFFER_SIZE 0

//#define MAX_CMD_SIZE 96
//#undef BUFSIZE
//#define BUFSIZE 16
</code>
</pre>
</details>

thinkyhead, 2019-02-25:
Please ZIP your configuration files together and drop the ZIP file on your next reply.

thinkyhead, 2019-02-25:
Where did you come by these configurations? We don't yet have AnyCubic support in Marlin 2.0.x.

thinkyhead, 2019-02-25:
My first suggestion is, do not define these in your `Configuration.h`. Comment out these lines…
```cpp
  //#define LEFT_PROBE_BED_POSITION MIN_PROBE_EDGE
  //#define RIGHT_PROBE_BED_POSITION (X_BED_SIZE - (MIN_PROBE_EDGE))
  //#define FRONT_PROBE_BED_POSITION MIN_PROBE_EDGE
  //#define BACK_PROBE_BED_POSITION (Y_BED_SIZE - (MIN_PROBE_EDGE))
```
…and remove any other definitions of these in your `Configuration.h`. The defaults calculated by `Conditionals_LCD.h` should be correct for all cases.

closed 2019-02-25 by thinkyhead