MarlinFirmware/Marlin #13254: Not compiling on MKS SGen 1.3

FelixHauser, 2019-02-24:
### Description

The Firmware does not compile for the MKS SGen v1.4

macOS
Platformio

### Steps to Reproduce

1. Copy/Replace the suggested Configuration.h and Configuration_adv.h (Anycubic Kossel)
2. Select correct board on configuration.h
3. Set LPC1768 as environment
4. Build for the LPC1768 as environment

**Expected behavior:** 

I expected to build a .bin file for me to upload

**Actual behavior:** 

Fails the build

#### Additional Information

I followed the instructions on the marlinfw.org site step by step, that´s why I am assuming this is an bug.

**Reported Errors**

These are the lines in red shown in the console after the failed build:

/Users/felix/.platformio/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h:161:8: error: expected '(' before 'co
nstexpr'

/Users/felix/.platformio/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h:204:8: error: expected '(' before 'co
nstexpr'

/Users/felix/.platformio/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h:238:3: error: expected '}' at end of 
input

/Users/felix/.platformio/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h:241:8: error: expected '(' before 'co
nstexpr'

/Users/felix/.platformio/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h:310:8: error: expected '(' before 'co
nstexpr'

*** [output/LPC1768/src/src/HAL/HAL_LPC1768/HAL.cpp.o] Error 1

/Users/felix/.platformio/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h:161:8: error: expected '(' before 'co
nstexpr'

/Users/felix/.platformio/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h:238:3: error: expected '}' at end of 
input

/Users/felix/.platformio/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h:241:8: error: expected '(' before 'co
nstexpr'

/Users/felix/.platformio/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h:310:8: error: expected '(' before 'co
nstexpr'

*** [output/LPC1768/src/src/HAL/HAL_LPC1768/HAL_spi.cpp.o] Error 1

boelle, 2019-02-24:
remember to attach 2 config files as zip file, maybe also include your platformio.ini in the zip file

FelixHauser, 2019-02-24:
As I mention, these are the two found on the examples folder

boelle, 2019-02-24:
doooh, laughs on me :-D

FelixHauser, 2019-02-24:
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2898231/Archive.zip)
just in case...

p3p, 2019-02-24:
@thinkyhead merged an api change for Serial ports, can you use the previous commit while this is being looked into.

https://github.com/MarlinFirmware/Marlin/archive/88cc1d1a31f7fe21f8a3796878d2ab22e2e66603.zip

FelixHauser, 2019-02-24:
thanks @p3p I'll look into it right away

p3p, 2019-02-24:
May have jumped to conclusions, I can't reproduce this issue using the configs supplied, make sure the platformio framework and libraries are all up to date.

I'm a bit confused with which examples you said you used as the MKS SGen does not have any (or a pins file), and the Anycubic Kossel is runs an 8bit AVR.

boelle, 2019-02-24:
if there is no pins file for the board it would also not have a board name to put in the the config files

make me wonder where the example or suggested config files mentioned in the first post comes from

boelle, 2019-02-24:
just gave it a quick look, marlin does not have any example configs for the anycubic kossel

only 2.0 have example configs for anycubic i3 mega, and that was me submitting those after i helped a user to get his printer going (done over 1000 comments on an issue)

p3p, 2019-02-24:
Anycubic Kossel is a delta so it's hidden in the delta sub folder.

boelle, 2019-02-24:
ahh, but yes BOARD_TRIGORILLA_14

same as the i3 mega, but given that there is no pins file for the MKS SGen it might be the reason why it does not compile

boelle, 2019-02-24:
hmmm

in the config its set to BOARD_MKS_SBASE, now i'm confused

p3p, 2019-02-24:
Indeed the config supplied compiled fine as it uses the MKS SBase which is why I mentioned making sure that platformio framework and libs where upto date.  The confusion was with everything else not existing including the SGen 1.3.

boelle, 2019-02-24:
i even google for SGen 1.3, no such board to be found anywhere

but there was an SBase 1.3 and Gen 1.3 but not SGen

vscode and platform io is my prefered choice over Arduino IDE as it updates libs etc automatic

only thing is to edit platformio.ini to correct build env

p3p, 2019-02-24:
But there is an MKS SGen 1.0 that will build with the LPC176x platform so I couldn't decide which part of the description was wrong, 

Even vscode needs told to update platformio, at least for me it does.

boelle, 2019-02-24:
i dont see it

[image]

FelixHauser, 2019-02-24:
I meant SBase, I just made a typo...
I will try to build in my other computer, in which I will have to install everything from scratch

boelle, 2019-02-24:
platformio.ini line 26

env_default = LPC1768

hmm.... just strange as it should actually build without errors

thinkyhead, 2019-02-24:
Hi! It looks like your issue is with your build environment. Please close this issue if switching computers solves the problem for you.

FelixHauser, 2019-02-24:
It's solved the problem. Sorry for the inconvenience. I hope at least it helps future issues.

closed 2019-02-24 by FelixHauser

boelle, 2019-02-24:
no problemo... we have all been there before :-D