MarlinFirmware/Marlin #13257: [BUG] Marlin 2.0.x I2C OLED display won't work on LPC1768

obertr0n, 2019-02-25:
### Description

Tried to configure an LPC1768 based mainboard (Biqu SKR V1.1) to use an 0.96 SSD1306 OLED display.
For connection I'm using P0.0 and P0.1 pins.

### Steps to Reproduce

1. [First Step]
In Configuration.h uncomment `#define U8GLIB_SSD1306`
In pins_BIQU_SKR_V1.1.h I have set the E1 pins to -1:
`#define E1_STEP_PIN        -1//P0_01
#define E1_DIR_PIN         -1//P0_00
#define E1_ENABLE_PIN      -1//P0_10`
2. [Second Step]
Connect OLED_SDA to P0_0 (DIR pin)
Connect OLED_SCL to P0_1 (STEP pin)

3. [Third Step]
Flash and reset

**Expected behavior:**
Display should operate properly and show all the info.
**Actual behavior:**
Display stays blank

#### Additional Information
I also tried to inspect the I2C bus with a cheap logic analyzer, but I'm not really familiar with SSD1306 chip.
I can provide traces from PulseView if necessary.

boelle, 2019-02-25:
where are the 2 config files ? put them here in a zip file

kAdonis, 2019-02-25:
Many of these OLEDs can be configured by solder jumpers, check if yours is configured to use I2C and not SPI and verify it is using I2C slave address 0x3C

obertr0n, 2019-02-25:
@kAdonis This is the display that I have, I did tested it with a bare Arduino code and it works OK over I2C.
[image]
@boelle I did attached the ZIP with the Config files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2902216/Configuration.zip)
And the pinout of the SKR
[SKR V1.1 PINOUT.pdf](https://github.com/MarlinFirmware/Marlin/files/2902218/SKR.V1.1.PINOUT.pdf)
+ pins.h file
[pins_BIQU_SKR_V1.1.h.zip](https://github.com/MarlinFirmware/Marlin/files/2902222/pins_BIQU_SKR_V1.1.h.zip)
Also, what is the purpose of these 2 files: HAL_LCD_I2C_routines.c and u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp they seem to address the same device, SSD1306..

obertr0n, 2019-02-26:
I had some time today to investigate more this problem.
What I have noticed is that the initialization sequence of the OLED is interrupted by ~40ms.
I couldn't figure out what exactly was causing this.
I also experienced something strange: when using USB power the logic analyzer shows the expected address 0x3C, but when using the power connector, the I2C address is 0x7F.
+ When powered by USB:
Delay between I2C init sequences:
[image]
First part:
[image]
Second Part:
[image]

+ When powered by external power supply:
[image]

Another strange behavior is that the board disconnects from PC and "freezes" after ~2-5mins.

obertr0n, 2019-02-28:
I have managed to get the display working, a standalone display + an encoder.
Here is the commit that is working with my setup:
https://github.com/obertr0n/Marlin/commit/b53a5bfd694c8bb8569efbb6b88cda306496938e

From what I could tell it works ok, to a point:
[image]
any idea what might be causing the bottom line to display like that?

kAdonis, 2019-02-28:
Great you got it working!
The bottom line seems to be the first line from "3D Printer Ready" which should be at the bottom not at the top
Your display is shifted

obertr0n, 2019-02-28:
I noticed that, what is strange is that it displays correctly for 1-2 mins and then that glitch is happening.
Really strange..

th33xitus, 2019-02-28:
@obertr0n are you using 5V or 3.3V as input voltage for the OLED?
I experienced kind of a similar strange behaviour with using 5V.
Using 3.3V instead works flawless.

obertr0n, 2019-02-28:
only power it, or also level shift sda/scl?

AnHardt, 2019-02-28:
Does the display return to normal when you switch to the menus and back, what should reinit the display?

obertr0n, 2019-02-28:
@kAdonis I tried powering the display from 3.3V. Unfortunately no change.

@AnHardt No..it displays every menu with that offset.

thinkyhead, 2019-03-01:
Getting closer…. I would suggest scrutinizing the LCD init sequence. Maybe there's something different about the command-set for this display…?

AnHardt, 2019-03-01:
The interesting part of the story is
> displays correctly for 1-2 mins and then that glitch is happening.

but reinitialising does not help. (Or don't we reinicialising any more?)

That makes a software problem unlikely.
Maybe interupted data transfer?

What are you doing in this 1-2 minutes?
Try to do nothing. No fan, no stepper, no heater activity - to minimize electrical noise and jitter in I2C timing.

Or can it be a heat problem? Also unlikely when powered and level-shifted to 3.3V. 
Can you feel a hot spot on the display/board?

What remains? Display-hardware fault?

obertr0n, 2019-03-01:
@AnHardt My setup is just the SKR board, the standalone OLED + encoder on a breadboard + a stepper driver and a stepper motor.
This happens every time, regardless of doing something on the board or not. 

If it was an interruption in the data transfer, shouldn't  it restore to normal, even for a small period when a new refresh is done?

I just tested without doing anything, without the stepper driver and motor. Just the display and the encoder, powered from external adapter (12V) after ~3:30 the glitch appeared.
No signs of heat or anything like that.

AnHardt, 2019-03-01:
Try the u8g-lib examples on the same hardware to find out if it's a Marlin problem or not.

obertr0n, 2019-03-01:
Just checked with a 128x32 display, no change in code and after some time the same happened.
I will still poke around to see if I can find something.

ShohninDmitriy, 2019-03-01:
It seems the problem is in the chip and the poor implementation of the I2C Protocol as even in the Clipper the same problem with hanging. And Smoothieware doesn't even support I2C in any way.

obertr0n, 2019-03-01:
@ShohninDmitriy If this would be the case I would've expected to have it as a known bug. Or maybe there are not that many users of this small OLED displays.

obertr0n, 2019-03-20:
Hi, it appears that I solved the issue, I don't really know if this is applicable for everyone, so I will leave my fork here for reference: https://github.com/obertr0n/Marlin/tree/marlin_2.0.x_ssd1306/Marlin
Affected files: 
 **HAL_LCD_I2C_routines.c**
**u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp**

closed 2019-03-28 by obertr0n

psavva, 2019-04-04:
@thinkyhead ,. Is it worth that you review the changes as per the last comment and see if this is a bug in Marlin?

istyszy, 2019-06-24:
[image]

Hello, i have an i2c OLED display, with marlin 2.0 (2019-06-24) + SKR V1.3, i use hardware i2c from E1. When the board starts everything is OK, display works as it should, but after 40 sec or so, the display crashes.
Did someone had this problem? I used the display with ramps+mega and never had any issues? I use u8glib_SH1160. (also tried U8GLIB_SSD1306 same result)

Johetan, 2020-02-24:
@obertr0n 
I tried it also on an mks sgenl. Same result. Also level shifted, 3.3v, do you already a have a solution for platform.io compile?

thinkyhead, 2020-02-26:
> see if this is a bug in Marlin?

I'm relying on the experts. I just clean up the place.

parsa-0x1, 2020-03-25:
Hello everybody.
i am not clever as much as you. i need a simple example to learn ssd1306 with Lpc1768 keil compiler. who can help?
i really need it. Ive been searched everywhere. and i am not so talented to change other micro controller code. please help me. thank you so much.

parsa-0x1, 2020-03-25:
can you send me a simple code about lpc1768 for ssd1306  with keil compiler??