MarlinFirmware/Marlin #13259: compile errors Marlin 2.0 bugfix 2.0.x 32bit using VCS/PlatformIO

splitn2, 2019-02-26:
Hi there , firstly I want to thank all the amazing efforts of the developers and contributors to Marlin, its an amazing piece of work.

I have just started out installing Marlin 2.0 bugfix 2.0.x using VCS and PlatformIO.Whilst I can compile a running binary, it at least boots and runs on my 32 bit board, I get a good number of compile errors even though I get a SUCCESS result.

The board I am using is a Cohesion 3D Remix, which is noted in boards.h and I have entered "BOARD_COHESION3D_REMIX" in configuration.h , and I have specified the "env_default = LPC1769" in the platformio.ini file.

So it compiles and I can upload the firmware.bin, the board boots and renames the file firmware.cur as per the way its meant to, and the LCD is displaying Marlin and I can interact via the encoder dial with the menus. So it seems a basic runner despite all the compiler errors - lots of warnings......

as far as I know PlatformIO and VCS are fully up to date and I have pulled the latest git of the Marlin 2.0 and the bugfix. I am running MAC OS Mojave .

Should I be worried about compiler errors - these warnings, they are beyond my paygrade to understand and resolve. As per one forum note I made sure VCS was installed into applications folder ,and tried clean re-install of VCS and PlatformIO and cleaned out pioenvs  , still behaves the same. 

Files below, any assistance would be appreciated. 
Thanks, Richard

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1769; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x > NXP LPC1769
HARDWARE: LPC1768 120MHz 31.80KB RAM (464KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(strict)
Collected 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.3.1

upload disk:  /Volumes/NO NAME/

Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/DebugMonitor_LPC1768.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/HAL.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/HAL_spi.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/HAL_timers.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/MarlinSerial.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/include/digipot_mcp4451_I2C_routines.c.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/main.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_flash.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_sdcard.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_I2C_routines.c.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_pin_routines.c.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/usb_serial.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/HAL_LPC1768/watchdog.cpp.o
In file included from Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/../core/macros.h:24,
from Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/MarlinConfigPre.h:32,
from Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/../../sd/cardreader.h:24,
from Marlin/src/HAL/HAL_LPC1768/main.cpp:17:
Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/../core/minmax.h: In instantiation of 'constexpr decltype ((lhs + rhs)) MIN(L, R) [with L = int; R = unsigned int; decltype ((lhs + rhs)) = unsigned int]':
Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/SanityCheck.h:1962:18:   required from here
Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/../core/minmax.h:35:20: warning: comparison of integer expressions of different signedness: 'const int' and 'const unsigned int' [-Wsign-compare]
return lhs < rhs ? lhs : rhs;
~~~~^~~~~
Compiling .pioenvs/LPC1769/src/src/HAL/shared/HAL_spi_L6470.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/shared/I2cEeprom.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/shared/SpiEeprom.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/shared/backtrace/backtrace.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/shared/backtrace/unwarm.cpp.o
Compiling .pioenvs/LPC1769/src/src/HAL/shared/backtrace/unwarm_arm.cpp.o
Marlin/src/HAL/shared/backtrace/unwarm_arm.cpp: In function 'UnwResult UnwStartArm(UnwState*)':
[... 1262 lines cut ...]
Environment DUE                 	[SKIP]
Environment DUE_USB             	[SKIP]
Environment DUE_debug           	[SKIP]
Environment LPC1768             	[SKIP]
Environment LPC1769             	[SUCCESS]
Environment melzi               	[SKIP]
Environment melzi_optiboot      	[SKIP]
Environment rambo               	[SKIP]
Environment sanguino_atmega644p 	[SKIP]
Environment sanguino_atmega1284p	[SKIP]
Environment STM32F1             	[SKIP]
Environment STM32F4             	[SKIP]
Environment ARMED               	[SKIP]
Environment mks_robin           	[SKIP]
Environment teensy35            	[SKIP]
Environment malyanm200          	[SKIP]
Environment esp32               	[SKIP]
Environment fysetc_f6_13        	[SKIP]
Environment linux_native        	[SKIP]
======================== [SUCCESS] Took 119.97 seconds ========================

Board name: BOARD_COHESION3D_REMIX
Build type: build
Environment used: LPC1769
2019-02-26 12:22:36.628519

p3p, 2019-02-26:
Warning are are just warnings that the compiler thinks something might be wrong and that somebody should check it's okay, in general programmers try to make sure that no warnings are detected during a build but sometimes the compiler can be very strict and things we know are not a problem are detected.

They will be cleaned up eventually and you can be sure they have been looked at to make sure that it not actually causing problems.

splitn2, 2019-02-26:
awesome, thanks very much @p3p , I'll keep going then and see where this leads me. I'm coming from a cnc background using mach3 grbl and smoothieware so Marlin pretty new , I like Marlin being able to configure so finely to your machine, what I am not enjoying is having to compile it but I guess thats how it rolls eh ! I see my VSC pulled some new bugfix this morning so I recompiled and the errors changed :) but I'll stress less and keep building. At some stage I may change to an alternative board but I hope to at least get started on the C3D board until I blow it up or out grow its IO but tbh it will prob do what I want for now.

cheers
Richard

closed 2019-02-26 by p3p