MarlinFirmware/Marlin #13263: [BUG] Marlin 2.0.x Error compile stm32f103ret

ShohninDmitriy, 2019-02-26:
Processing STM32F1 (platform: ststm32@<4.4.0; board: genericSTM32F103RE; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz 64KB RAM (512KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.3.1
|   |-- <SPI> 1.0
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD> #c8ac22f
|-- <SailfishRGB_LED> #2426fa2
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster> #3a18be5
|-- <Wire> 1.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <SPI> 1.0
Compiling .pioenvs\STM32F1\src\src\HAL\HAL_STM32F1\HAL.cpp.o
Compiling .pioenvs\STM32F1\src\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp.o
Compiling .pioenvs\STM32F1\src\src\HAL\HAL_STM32F1\HAL_sdio_STM32F1.cpp.o
Compiling .pioenvs\STM32F1\src\src\HAL\HAL_STM32F1\HAL_spi_STM32F1.cpp.o
In file included from c:\users\dima\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdlib.h:19,
from c:\users\dima\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\cstdlib:75,
from c:\users\dima\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\stdlib.h:36,
from C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\cores\maple/wirish.h:41,
from C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\cores\maple/Arduino.h:30,
from Marlin\src\HAL\HAL_STM32F1\HAL.h:41,
from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:34:
C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\system\libmaple/stm32f1/include/series/gpio.h:487:15: error: duplicate 'inline'
static inline __always_inline void afio_exti_select(exti_num exti, exti_cfg port) {
^~~~~~~~~~~~~~~
C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\system\libmaple\include/libmaple/nvic.h:112:15: error: duplicate 'inline'
static inline __always_inline void nvic_globalirq_enable() {
^~~~~~~~~~~~~~~
C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\system\libmaple\include/libmaple/nvic.h:119:15: error: duplicate 'inline'
static inline __always_inline void nvic_globalirq_disable() {
^~~~~~~~~~~~~~~
C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\cores\maple/ext_interrupts.h:109:15: error: duplicate 'inline'
static inline __always_inline void interrupts() {
^~~~~~~~~~~~~~~
C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\cores\maple/ext_interrupts.h:123:15: error: duplicate 'inline'
static inline __always_inline void noInterrupts() {
^~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\STM32F1\src\src\HAL\HAL_STM32F1\HAL.cpp.o] Error 1
In file included from c:\users\dima\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\stdlib.h:19,
from c:\users\dima\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\cstdlib:75,
from c:\users\dima\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\8.2.1\stdlib.h:36,
from C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\cores\maple/wirish.h:41,
from C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\cores\maple/Arduino.h:30,
from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/HAL_STM32F1/HAL.h:41,
from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
from Marlin\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp:26:
C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\system\libmaple/stm32f1/include/series/gpio.h:487:15: error: duplicate 'inline'
static inline __always_inline void afio_exti_select(exti_num exti, exti_cfg port) {
^~~~~~~~~~~~~~~
C:\users\dima\.platformio\packages\framework-arduinoststm32@2.1.180219\STM32F1\system\libmaple\include/libmaple/nvic.h:112:15: error: duplicate 'inline'
static inline __always_inline void nvic_globalirq_enable() {
^~~~~~~~~~~~~~~
[... 61 lines cut ...]
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [SKIP]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [ERROR]
Environment STM32F4                     [SKIP]
Environment ARMED                       [SKIP]
Environment mks_robin                   [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment fysetc_f6_13                [SKIP]
Environment linux_native                [SKIP]
===================================================================== [ERROR] Took 8.04 seconds =====================================================================
The terminal process terminated with exit code: 1

megamih, 2019-03-02:
Hi @ShohninDmitriy ,

Looks like I have similar issue with mks_robin (platform: ststm32; board: genericSTM32F103ZE; framework: arduino). And for me issue is in toolchain-gccarmnoneeabi package version "1.80201.190214". I replaced it with, I believe, previous version toolchain-gccarmnoneeabi-windows-1.70201.0 from https://dl.bintray.com/platformio/dl-packages/ - and it compiles.

Best regards,
Alex

ShohninDmitriy, 2019-03-03:
That's right, replaced the compiler and everything compiled, thank you very much.

jyzhkj, 2019-03-03:
it's can't use for me.
[env:genericSTM32F103RE]
platform = ststm32
board = genericSTM32F103RE
framework = arduino

ShohninDmitriy, 2019-03-03:
Building .pioenvs\STM32F103RET6\firmware.bin
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  21.8% (used 14272 bytes from 65536 bytes)
PROGRAM: [===       ]  28.9% (used 151476 bytes from 524288 bytes)
================================================================= [SUCCESS] Took 256.85 seconds =================================================================

=========================================================================== [SUMMARY] ===========================================================================
Environment STM32F1                     [SKIP]
Environment STM32F103RET6               [SUCCESS]
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]

closed 2019-03-03 by thinkyhead

comps, 2019-03-27:
I'm still getting this with today's `bugfix-2.0.x`.

Grepping around, there are only a few places that define `__always_inline` and STM32F1 seems to just be different:
```
$ grep 'define \+__always_inline' -R ~/.platformio/ .pio* Marlin/
/home/user/.platformio/packages/framework-arduinoststm32/STM32F4/cores/maple/libmaple/libmaple_types.h:  #define __always_inline inline __attribute__((always_inline))
/home/user/.platformio/packages/framework-arduinoststm32/STM32F1/system/libmaple/include/libmaple/libmaple_types.h:#define __always_inline __attribute__((always_inline))
/home/user/.platformio/packages/framework-arduinoststm32/STM32F3/cores/maple/libmaple/include/libmaple/libmaple_types.h:#define __always_inline inline __attribute__((always_inline))
Marlin/src/HAL/HAL_DUE/usb/compiler.h:#   define __always_inline   __forceinline
Marlin/src/HAL/HAL_DUE/usb/compiler.h:# define __always_inline   inline __attribute__((__always_inline__))
Marlin/src/HAL/HAL_DUE/usb/compiler.h:# define __always_inline   _Pragma("inline=forced")
```
However even if the definition leaks from elsewhere (due to `ifndef`), I can't seem to figure out from where - modifying all the other definitions doesn't change the error one bit.