MarlinFirmware/Marlin #13264: [BUG] MARLIN 2.0 G28 in sliced Gcode file disregards Z-min

dystechnic, 2019-02-26:
Hi,
A G28 issued from a terminal (octopi) results in a clean and good home for X,Y and Z. endstops are triggerd and the printer acts as expected.
But when I have a sliced Gcode filwith a G28 command in it, the X and Y homing is succesfull but Z-homing is not. The Z-min endstop is a inductive sensor, triggers fine but is ignored by the printer. 

The result is the head slamming into the table. While the expected result would be a clean homing of the Z axis.

To clarify: 
This results in a headslam:
_;FLAVOR:Marlin
;TIME:3797
;Filament used: 1.05236m, 0m, 0m
;Layer height: 0.1
;Generated with Cura_SteamEngine master
T0
M140 S55
M105
M190 S55
M104 S215
M105
M109 S215
M82 
G28 
G1 Z15.0 F6000 
G92 E0
G1 F200 E3
G92 E0
M83 ;relative extrusion mode
G1 F1800 E-7
;LAYER_COUNT:199
;LAYER:0_

This works, but I have to do a manual G28 before printing:
_;FLAVOR:Marlin
;TIME:3797
;Filament used: 1.05236m, 0m, 0m
;Layer height: 0.1
;Generated with Cura_SteamEngine master
T0
G0 F3600 X0 Y0 Z10
M190 S55
M109 S210
M82
G92 E0
G1 F200 E3
G92 E0
M83 ;relative extrusion mode
G1 F1800 E-7
;LAYER_COUNT:199
;LAYER:0_

Attached are my config files.

[configuration.h](https://ghostbin.com/paste/dq7pg)
[configuration_adv.h](https://ghostbin.com/paste/mukr2l)

reloxx13, 2019-02-27:
when u did flash?

there was a bug which leet to smashing a few days/1-2weeks ago.

update ur marlin sourcecode

reloxx13, 2019-02-27:
this was it 😵 

https://github.com/MarlinFirmware/Marlin/pull/13067

dystechnic, 2019-02-28:
I checked out my version on the 23th of februari. The fix you mentioned is already in the version I use.

I noticed yesterday that the problem I have also occurs if I home the Z-axes from Octoprint after a print is finished and before I start a new one.
I've found a setting in configuration_adv.h 'ENDSTOPS_ALWAYS_ON_DEFAULT' not sure if this will solve the issue if I enable it, but I guess I could try this.

thinkyhead, 2019-03-03:
Do you have your Z homing configured to probe once or twice? Do you have a Z raise configured for homing, high enough so that the probe is not already triggered when starting G28? This is important because these probes don't show up as continually triggered. They trigger when their state changes.

dystechnic, 2019-03-05:
I have set #define Z_HOMING_HEIGHT to 5  and #define MULTIPLE_PROBING to 3. 
But the head still slams into the bed. The probe isn't trggered before the G28 command and triggers at the right moment but doesn't stop.

dystechnic, 2019-03-06:
I did some more tests and noticed something that might be usefull to find out what is happening. 
When I home all axes with G28 the head is inthe center of the bed and the inductive probe is triggerd according to the indicator light. But the M119 command tells me that the Z-probe is not triggered.
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open
Recv: ok P31 B7

If I lower the bed (after a G28) and manually force the probe to trigger it M119 show it is triggered for a short time and the changes to it to not triggered.
A M120 (which is as far as I know effectively the same as ENDSTOPS_ALWAYS_ON_DEFAULT) does not change this behavior
Following is a short debugging log of what happens when I start a printjob and the bed crashes into the head. Note, the head is 5mm above the bed before the job starts and a manual homing is done already.

Send: N8 M82*17
Recv: ok N8 P31 B7
Send: N9 G28*26
Recv: >>> G28
Recv: Machine Type: Core
Recv: Probe: FIX_MOUNTED_PROBE
Recv: Probe Offset X:10 Y:10 Z:-0.10 (Right-Back & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR
Recv:  (disabled)
Recv: 
Recv:   current_position=(0.00, 0.00, 5.00) : bracket_probe_move
Recv: > endstops.enable(true)
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv: >>> do_blocking_move_to(-555.00, -600.00, 5.00)
Recv: <<< do_blocking_move_to
Recv: >>> homeaxis(X)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(X, -555.00, [50.00])
Recv: <<< do_homing_move(X)
Recv: Move Away:
Recv: >>> do_homing_move(X, 5.00, [50.00])
Recv:  T:216.67 /215.00 B:55.03 /55.00 @:24 B@:22
Recv: <<< do_homing_move(X)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(X, -10.00, 25.00)
Recv: <<< do_homing_move(X)
Recv: >>> set_axis_is_at_home(X)
Recv: For X axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv: > home_offset[X] = 0.00
Recv:   current_position=(0.00, 0.00, 5.00) :
Recv: <<< set_axis_is_at_home(X)
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Y, -600.00, [50.00])
Recv: <<< do_homing_move(Y)
Recv: Move Away:
Recv: >>> do_homing_move(Y, 5.00, [50.00])
Recv: <<< do_homing_move(Y)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Y, -10.00, 25.00)
Recv: <<< do_homing_move(Y)
Recv: >>> set_axis_is_at_home(Y)
Recv: For Y axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv: > home_offset[Y] = 0.00
Recv:   current_position=(0.00, 0.00, 5.00) :
Recv: <<< set_axis_is_at_home(Y)
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Y)
Recv: Z_SAFE_HOMING >>>
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv:   destination=(175.00, 190.00, 5.00) : Z_SAFE_HOMING
Recv: >>> do_blocking_move_to(175.00, 190.00, 5.00)
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:216.67 /215.00 B:55.01 /55.00 @:27 B@:24
Recv: >>> homeaxis(Z)
Recv:   current_position=(175.00, 190.00, 5.00) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(10.00)
Recv: >>> do_blocking_move_to(175.00, 190.00, 10.10)
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:216.67 /215.00 B:55.01 /55.00 @:28 B@:24
Recv: >>> do_blocking_move_to(175.00, 190.00, 10.10)
Recv: <<< do_blocking_move_to
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Z, -525.00, [4.00])
Recv: echo:busy: processing
Recv:  T:216.04 /215.00 B:55.01 /55.00 @:38 B@:24
Recv: echo:busy: processing
Recv:  T:215.83 /215.00 B:55.01 /55.00 @:39 B@:23
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2605

The unexpected error is me hitting the reset button of the controller when the head has hit the bed to prevent further damage.

dystechnic, 2019-03-06:
In addition to the former post I noticed that in Bugfix 2.0.x the state of the Z-endstop changes when I query it's state I noticed the following behaviour:

Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: TRIGGERED
Recv: ok P31 B7
Recv:  T:23.12 /0.00 B:22.32 /0.00 @:0 B@:0
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: TRIGGERED
Recv: ok P31 B7
Recv:  T:23.12 /0.00 B:22.43 /0.00 @:0 B@:0
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open
Recv: ok P31 B7
Recv:  T:23.12 /0.00 B:22.46 /0.00 @:0 B@:0
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open
Recv: ok P31 B7
Recv:  T:23.12 /0.00 B:22.46 /0.00 @:0 B@:0
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: TRIGGERED
Recv: ok P31 B7
Recv:  T:23.12 /0.00 B:22.46 /0.00 @:0 B@:0
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: TRIGGERED
Recv: ok P31 B7
Recv:  T:23.12 /0.00 B:22.50 /0.00 @:0 B@:0
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open
Recv: ok P31 B7

The bed did not move and the probe indicated that is was triggered the whole time. Enabling Endstop Noise Threshold did not help. I tried different values there, but nothing changed.

I tried it with a different firmware (MK4DUO V4.3.6  and the result was as I expected it to be:

Send: N129 M119*32
Recv: Reporting endstop status
Recv: x_min: NOT TRIGGERED
Recv: y_min: NOT TRIGGERED
Recv: z_min: TRIGGERED
Recv: ok N129 P31 B7
Send: N130 G4 S2*110
Recv: ok N130 P31 B7
Send: N131 M119*41
Recv: Reporting endstop status
Recv: x_min: NOT TRIGGERED
Recv: y_min: NOT TRIGGERED
Recv: z_min: TRIGGERED
Recv: ok N131 P31 B7
Send: N132 G4 S2*108
Recv: ok N132 P31 B7
Send: N133 M119*43
Recv: Reporting endstop status
Recv: x_min: NOT TRIGGERED
Recv: y_min: NOT TRIGGERED
Recv: z_min: TRIGGERED
Recv: ok N133 P31 B7
Send: N134 G4 S2*106
Recv: ok N134 P31 B7
Send: N135 M119*45
Recv: Reporting endstop status
Recv: x_min: NOT TRIGGERED
Recv: y_min: NOT TRIGGERED
Recv: z_min: TRIGGERED
Recv: ok N135 P31 B7

dystechnic, 2019-03-07:
I've solved my problem!

Triggered by the differences between Marlin and MK4DUO I turned to the electronics surrounding my LJ12A3-4-Z/BX Inductive Proximity Sensor. I put an oscilloscope to the z-min pin and saw that the signal levels where slightly off. Sometimes within and sometimes outside the logic boundaries. After some research I discovered that this type of probe has an internat 10K pull-up resistor. So, the voltage divider I use to convert the 12V signal back to 5V did not work as expected.
I did some calculations and decided that a single resistor in combination with the internal pull-up should do the trick. And it did! 
Later I discovered a website from someone who did the same. https://mertarauh.com/2017/01/18/dont-trust-the-internet-and-how-to-add-an-inductive-proximity-sensor-to-your-3d-printer-the-proper-and-easiest-way/ He has some additional ways to connect the sensor. But I went for one 6.8K resistor.
No more problems and a clean logic 5V and 0V from the sensor.

Thank you all for trying to solve this issue with me.

closed 2019-03-07 by dystechnic

thinkyhead, 2019-03-08:
Good news. Inductive sensors can be fun! Capacitive sensors have started to take over around here…

InsanityAutomation, 2019-03-08:
Any 10-30vdc sensor can essentially do the same, including capacitive. Isolation is recommended in these scenarios to be 100% safe from sending high voltage right into the input.

Roxy-3D, 2019-03-08:
> Good news. Inductive sensors can be fun! Capacitive sensors have started to take over around here…

My opinion is micro-switches are more reliable and easier to get working...   But to each his own.