MarlinFirmware/Marlin #13265: [FR] (feature request title)

jmdearras, 2019-02-26:
### Description

Need a way to increase the case light PWM frequency.  If dimmed any, the flicker is horrible

### Steps to Reproduce

1. enable case light on pwm pin with LED array
2. turn case light on
3. Note at 255, light is solid, but at 127, severe flicker.

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

jmdearras, 2019-02-28:
I would be happy to add it, if I could figure out how the softpwm works for the case light.

p3p, 2019-02-28:
The LPC176x framework Marlin uses doesn't currently support changing the period of PWM, all the channels use the same hardware timer and servos require it to be 20ms (50Hz), I didn't notice any flicker on the board LEDs when testing it,

jmdearras, 2019-02-28:
OK.  50hz is very visible for case lights.  Any hopes for supporting it in the future?

7eggert, 2019-03-04:
There is a slim hope that I won't abandon my half-baked idea of supporting I²C PWM pins. (PCA9685).

Andrei-Pozolotin, 2019-03-05:
+1. the flicker is severe @ 50hz

Bob-the-Kuhn, 2019-05-01:
I'm wondering what a non-flicker PWM frequency would be.

Please try this file.  It sets all PWMs to 60Hz.

It replaces the one in the following directory:
C:\Users\YOUR_USER_NAME\.platformio\packages\framework-arduino-lpc176x\cores\arduino

[pwm.zip](https://github.com/MarlinFirmware/Marlin/files/3135173/pwm.zip)

If 60Hz still flickers then edit line 23 in the file.

jmdearras, 2019-05-02:
Yes, that essentially removed the flicker.  Does it have any side effects?

ManuelMcLure, 2019-05-02:
I'm sort of wondering whether we can take advantage of the special attributes of a servo signal to make this work better. Servos expect signals every 20ms or so (50Hz) _but_ said signal only has a valid range of 0-2ms. So it's going to be off for at least 9/10ths of the period.
Suppose we set the PWM frequency to 500Hz and have code to disable the output pin for 9 out of the 10 cycles on pins defined as servo outputs. This would allow the maximum resolution on the servo while also allowing for higher frequency output for LEDs. We'd probably need an ISR that gets triggered at each reset of the timer and would connect/disconnect the pin from the PWM on a 1:9 cycle.
This wouldn't fix the fan problem, but would make LEDs work better. And it could add to the servo PWM resolution, unless I'm misunderstanding how PWM for servos works.

jmdearras, 2019-05-02:
Answered my own question.  It can't read the SD card.

Bob-the-Kuhn, 2019-05-04:
@jmdearras - most unexpected.  Changing just the PWM file shouldn't affect SD card access.  I'll have to investigate.

@ManuelMcLure - the servo signals are currently handled by the CPU's PWM module.  This keeps the jitter to almost zero.  Unfortunately it can't do what you proposed.  I'm wondering if setting all servos to 60Hz would work.  I know the BLTouch will operate properly up to about 500Hz.

ManuelMcLure, 2019-05-04:
Wouldn't changing the ``PWMENAx`` bit in the ``PCR`` register at each timer reset work?

Bob-the-Kuhn, 2019-05-04:
That would require using an ISR.  If we're using an ISR then there's easier ways of doing it.

ManuelMcLure, 2019-05-04:
It would require an ISR, but the ISR would only trigger 500 times a second, and would be very simple, something like this:
```
servo_pin_match_isr()
{
  static byte cycle = 0;
  cycle = ++cycle % 10;
  switch (cycle)
  {
  case 0:
    // Enable servo pins
    break;
  case 1:
    // Disable servo pins
    break;
  }
}
```
We'd still use the CPU to generate the length of the pulse so we should get the same even pulse length, and because we're mapping the 0-255 PWM range to the 0-2ms pulse we should get the maximum resolution on the servo pin.

Bob-the-Kuhn, 2019-05-04:
The problem with ISRs is other ISRs cause jitter in the start position (period) and in the pulse width.  They are marginal for devices like the BLTouch.

ManuelMcLure, 2019-05-04:
That's why I'm thinking of attaching/detaching the pin from the PWM right when the pulse ends. At this point the timer's already turned the output off and we have the time between pulses to attach/detach the pin. This might have trouble right when we do a 2ms pulse, but that can be resolved by tweaking the PWM frequency - if instead of 500Hz we go to 450Hz we'll have an extra 2ms available to connect/disconnect the pin before the next pulse starts, with minimal loss in resolution. The pulses are still generated by the hardware PWM so there should be no jitter, we're just ignoring 9 out of 10 of them.

ManuelMcLure, 2019-05-04:
Correction - 450Hz would give us an extra 200us to disconnect the pin between pulses.

ej0rge, 2019-05-05:
Seeing the same issue here having upgraded a ramps board with a re-arm. Case lighting is a strobe at less than 255. 

Is software pwm for illumination not a realistic option?

Bob-the-Kuhn, 2019-05-05:
The problem is not enough timers on the LPC1768 to give different periods to all the different types of PWM users.

We're looking at offering a choice of 50Hz or 60Hz for the PWMs.   See PR #13909.

It could be quite some time before that PR is implemented.  Until then please use the **pwm.zip** I posted four days ago in this issue.

ej0rge, 2019-05-05:
OK, I may look into that. I specifically want to use both pwm chassis lighting and a bltouch, so, I guess it turns out to be a slightly demanding configuration.

jmdearras, 2019-05-08:
> Answered my own question. It can't read the SD card.

This was my mistake.  I did not notice the changes in selecting sd card location.  Does this mean I can use 60hz on a SKRV1.1 with no BLTouch, with a case light, and 50hz with BLTouch and no case light?

UnderXirox, 2019-05-14:
Bonjour voici une doc utils pour les mks sbase
[https://github.com/makerbase-mks/MKS-SBASE/blob/master/Hardware/MKS%20SBASE%20V1.3_002%20SCH.pdf](url)

Bob-the-Kuhn, 2019-05-14:
I love schematics.  

Thanks

---

J'aime les schémas.

Merci

boelle, 2019-06-22:
missing a label

thinkyhead, 2019-07-11:
Please relocate discussion to #13909.

closed 2019-07-11 by thinkyhead