MarlinFirmware/Marlin #13268: Marlin 2.0 AD595 Arduino Due

svs072477, 2019-02-27:
Marlin 2.0 Thermocouple k-type with AD595 RURAMPS + Arduino Due

When the temperature increases, the readings issued by the firmware, on the contrary, decrease. On Repetier Firmware works correctly

boelle, 2019-02-27:
where are your 2 config files attached as zip?

svs072477, 2019-02-27:
OK, I will add configs in the evening

AnHardt, 2019-02-27:
Thermocouples are polar. Swap the connectors.

svs072477, 2019-02-27:
Yes, indeed. Just when was the Repetier to the polarity readings are not dependent, or I was always hooked up correctly :)

boelle, 2019-03-06:
so this is still an issue? just strange

svs072477, 2019-03-06:
No, I Swap the connectors, and issue solved

boelle, 2019-03-06:
@svs072477 then give the close button below a friendly thought :-D

svs072477, 2019-03-06:
Oh, sorry

closed 2019-03-06 by svs072477