MarlinFirmware/Marlin #13271: [FR] Add Slice Engineering High Temp Thermistor

SliceEngineering, 2019-02-28:
### Description

Slice Engineering builds a high temperature (up to 450C) thermistor that still gives accurate readings at room temperature. Please add the attached ADC table to allow users to select this thermistor as an option in their firmware. 

### Steps to Reproduce

Including Slice high temp thermistor will require:

1. Adding attached ADC table to Marlin
2. Allowing users to select the Slice Engineering High Temp Thermistor from their firmware settings

#### Additional Information
Video from Chris Riley demonstrating updating the firmware: https://www.youtube.com/watch?v=qwboQDMY-h8&t=2s

[Slice 450° C thermistor - RT table with ADC - Sheet1.pdf](https://github.com/MarlinFirmware/Marlin/files/2913459/Slice.450.C.thermistor.-.RT.table.with.ADC.-.Sheet1.pdf)

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-02-28:
best is to fork bugfix 2.0, then add the thermistor table and submit a PR to bugfix 2.0

it can be included much faster that way

@thinkyhead agree?

SliceEngineering, 2019-02-28:
I'm fine with whatever you think is best. Just let me know what I need to do.

boelle, 2019-02-28:
I think @thinkyhead would prefer a PR against Bugfix 2.0

InsanityAutomation, 2019-02-28:
Make a water cooling option I can buy! That was the deciding factor for me to the Dyze! 

Ive added these before and can work with you making the PR if youre not familiar with the process. Its fairly quick and painless. All the information needed is in the video by the look of it, but if you have the thermistor 67 file he made, I can finish it in 20 minutes. Otherwise, add a few minutes comparing from the video!

The values below 0 I dont recommend including all of. That tight ADC reading compared to large temp change wont give repeatable results, and it isnt really necessary down there anyway. 

To simplify making that temp table in the future, I updated a spreadsheet tool on thingiverse to the format used by Marlin 2.0 https://www.thingiverse.com/thing:3300383/files so it may save some time in the future!

boelle, 2019-02-28:
@InsanityAutomation would the excel sheet they attached in first post do the trick?

InsanityAutomation, 2019-02-28:
> @InsanityAutomation would the excel sheet they attached in first post do the trick?

All the data is there, so it can be used to populate what I had. Either way, I just did it and ill open the PR momentarily.

boelle, 2019-02-28:
another one bites the dust (ie issue solved) :-D

InsanityAutomation, 2019-02-28:
Yeah, I did another one of these recently so it was all fresh in my mind. Figured id throw the few minutes at it lol

SliceEngineering, 2019-02-28:
Whoa! That was quick! Thanks guys! 

Just to clarify for this mechanical guy, this means that to use our thermistor, a Marlin user just needs to download the latest bug fix, correct? 

@insanityautomation you're killing me with the Dyze thing! We've got liquid cooling in the pipeline.

boelle, 2019-02-28:
@SliceEngineering yep user just have to select thermistor number 67

once the PR has been merged that is, and yes bugfix 2.0

InsanityAutomation, 2019-02-28:
@SliceEngineering it will be as soon as Scott clicks the merge button! He usually goes through these in the evenings so it will most likely be there by tomorrow unless he has some issue or I made a typo somewhere :-P

Any eta on water-cooling stuff so I know when to poke back?

boelle, 2019-02-28:
while we are on water cooling, have anyone ever fitted some waterblock on the steppers? or is that to shoot birds with canon's ?

InsanityAutomation, 2019-02-28:
@boelle same company that's a bad word to these guys mentioned above... and it becomes absolutely necessary with chamber temps high enough to run ultem. forced air cooling is also far less effective against the issue with just a fan mentioned above.

boelle, 2019-02-28:
reason it came in to my head is that most hotends have this stupid little fan to cool the heatsink, and the smaller they get the more they sound like a jet engine at full blast with afterburner on

so i thought that if going to play with watercooling why not put the steppers in the loop while at it

InsanityAutomation, 2019-02-28:
> reason it came in to my head is that most hotends have this stupid little fan to cool the heatsink, and the smaller they get the more they sound like a jet engine at full blast with afterburner on
> 
> so i thought that if going to play with watercooling why not put the steppers in the loop while at it

Absolutely. If theyre in the chamber anyway. When running a 120c chamber, most steppers wont really work for long anyway without something.... The old nasa lulzbot article they used compressed air boxes around em. Thats why forced air is no longer effective for the hotend as well. Im sure the guys @SliceEngineering know all this better than I do since theyve designed a hotend for those materials though!

SliceEngineering, 2019-03-01:
@boelle great! Thanks for the explanation. Is there a way for me to know definitively when the merge has been completed?

RE: water cooling. We don't have a publicly announced date for launch yet, unfortunately. Mosquito does actually make it possible to print Ultem/PEI/PEEK in a heated chamber with only air cooling - heat break is that good. We have some medical device and automotive customers doing just that. The ideal of course would still be to use liquid cooling for the stepper motors.

InsanityAutomation, 2019-03-01:
@SliceEngineering It was merged last night so all good!

I'm in the automotive category there. I'm a controls engineer.

InsanityAutomation, 2019-03-01:
@SliceEngineering side note - You guys going to be at RAPID this year?

SliceEngineering, 2019-03-01:
Fantastic! Thanks for the quick work guys. I'll announce the compatibility on the socials today. 

Awesome! I'd love to talk automotive :) We will be at RAPID - sharing some table space with Matterhackers

InsanityAutomation, 2019-03-01:
@SliceEngineering Then Ill see you there im sure, Im only 20 minutes away this year so itd be pretty bad if I didnt make it lol

SliceEngineering, 2019-03-02:
@InsanityAutomation awesome! Please come by and introduce yourself!

boelle, 2019-03-03:
@SliceEngineering since this one is done maybe consider clicking close below?

closed 2019-03-03 by SliceEngineering

InsanityAutomation, 2019-03-18:
@SliceEngineering Ive gotten some reports that the bottom end of the table is quite off from the data shown above. Any way we could get ahold of some physically validated data? If not, will you have one at MRRF, and I can at least get close with my FLIR?

SliceEngineering, 2019-03-20:
@InsanityAutomation we have validated the data both internally and through
a 3rd party. I would be interested in talking to anyone that has
encountered issues with the table. We will be at MRRF and have our own FLIR
set up, likely rigged to a printer.

Stay zesty,

Dan

Daniel Barousse
Co-founder
Slice Engineering, LLC
www.sliceengineering.com

InsanityAutomation, 2019-03-20:
@SliceEngineering sounds good, may be a wiring, board, or other issue. I'll see you there, and should be pretty easy to confirm either way!