MarlinFirmware/Marlin #13273: multiple filament runout sensors

mitchcecg, 2019-02-28:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

mitchcecg, 2019-02-28:
I hope this is the correct place for this question if not please direct me to the correct place.

I have 3 extruders and I have 3 filament runout sensors that I can connect to pins on my rumba board. the question is does Marlin handle this and how to configure Marlin for this. I only see a way to have one filament runout sensor 

Thanx in advance for the help

FanDjango, 2019-02-28:
If they are three switches of the NC (normally closed type), you could wire them in series and consider them to be "one" switch. This was my situation.

If they are NO (normally open) you could wire them in parallel and consider them to be a single switch.

Sensors of the electronic type are typically open drain/open collector pulling to ground when triggered, these are like NO switches and can be wired in parallel (that's a wired-OR).

With both methods, of course you have no way to determine WHICH filament has run out - you need to determine that manually in such a case.

boelle, 2019-02-28:
>  for this question 
> 
the correct place for questions are discord and facebook

InsanityAutomation, 2019-02-28:
Also, for reference in Marlin 2.0 configuration.h   #define NUM_RUNOUT_SENSORS   3

And 
#define FIL_RUNOUT_PIN     xx
#define FIL_RUNOUT2_PIN    xx
#define FIL_RUNOUT3_PIN     xx

closed 2019-03-03 by thinkyhead