MarlinFirmware/Marlin #13274: [BUG] MKS SBASE: expose controller sd card to host as mass storage device

Andrei-Pozolotin, 2019-02-28:
1. **PROBLEM**
here: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_MKS_SBASE.h#L157
a combination of options:
```
#define USB_SD_ONBOARD
#define LPC_SD_ONBOARD
```
is supposed to expose controller sd card to host as mass storage device,
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_MKS_SBASE.h#L221
which does not work, due to
```
  #define SS_PIN           P0_06   // Chip select for SD card used by Marlin
  #define ONBOARD_SD_CS    P0_06   // Chip select for "System" SD card
```

2. **SOLUTION**
borrowing similar snippet from
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_BIGTREE_SKR_V1.3.h#L213
```
  #define SS_PIN           P0_16   // Chip select for SD card used by Marlin
  #define ONBOARD_SD_CS    P0_06   // Chip select for "System" SD card
```
now makes it work for MKS SBASE: sd card does show up on host as block device

3. working configuration snapshot
https://github.com/random-builder/Marlin/commit/2e4ab83204541d2a7b214feb1b2eac490b302ef7

p3p, 2019-02-28:
I have an MKS SBase myself and the current pins file is correct, the SKR pins file is the one with the error and has a PR to fix it. 

see the comments on #13267.

Andrei-Pozolotin, 2019-02-28:
well, my experience is different:
here
https://github.com/random-builder/Marlin/blob/2e4ab83204541d2a7b214feb1b2eac490b302ef7/Marlin/src/pins/pins_MKS_SBASE.h#L225
with this setup
```
  #define SS_PIN           P0_16   // Chip select for SD card used by Marlin
  #define ONBOARD_SD_CS    P0_06   // Chip select for "System" SD card
```
I can see sd disk on host
```
pi@octopi2:~ $ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sdb           8:16   1 14.9G  0 disk 
└─sdb1        8:17   1 14.9G  0 part 
mmcblk0     179:0    0 14.9G  0 disk 
├─mmcblk0p1 179:1    0 43.1M  0 part /boot
└─mmcblk0p2 179:2    0 14.8G  0 part /
```

while with the following:
```
  #define SS_PIN           P0_06   // Chip select for SD card used by Marlin
  #define ONBOARD_SD_CS    P0_06   // Chip select for "System" SD card
```
the card is gone:
```
pi@octopi2:~ $ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 14.9G  0 disk 
├─mmcblk0p1 179:1    0 43.1M  0 part /boot
└─mmcblk0p2 179:2    0 14.8G  0 part /
```

p3p, 2019-02-28:
With the default configs Marlin is sharing the onboard sdcard with the host, both cannot access the card at the same time, if Marlin has the card mounted it will not be available to the host. 

Try with the default configs and make sure Marlin does not have the card mounted.

Andrei-Pozolotin, 2019-02-28:
Ah, started to make sense, thank you. With my "fix", I am getting:
```
Send: M21
Recv: echo:SD init fail
```
Which setting tells Marlin NOT to mount sd card on boot?

gloomyandy, 2019-02-28:
Do you have a display defined? If not then Marlin will always mount the SD card on boot. You can use M22 to unmount the card. If you have a display defined then with the default pins settings Marlin will not mount the card on boot (and so it will be available to your PC). It is possible to change how Marlin works without a display be you will need to change the Marlin startup code to do that. But let's make sure that we understand the problem first.

Andrei-Pozolotin, 2019-02-28:
No display, using TFT on AUX1. Thank you for the hint. The task at hand is to find an easy way for quick remote re-flash of `mks-sbase/marlin` firmware via usb-connected `octopi`.

gloomyandy, 2019-02-28:
So are you happy with using m22? If not you can try commenting out this line...
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/Marlin.cpp#L1079
which I think will stop Marlin from mounting the SD card on boot.

Andrei-Pozolotin, 2019-02-28:
1. M22 works fine, thank you.

2. is there analog of smoothieware `on_boot.gcode`, see: http://smoothieware.org/sd-card

3. would it make sense to expose "sd auto start" behavior, i.e. convert from:
```
// Marlin.cpp
  #if ENABLED(SDSUPPORT) && DISABLED(ULTRA_LCD)
    card.beginautostart();
  #endif
```
into user-controlled setting:
```
// Marlin.cpp
  #if ENABLED(SDAUTOSTART)
    card.beginautostart();
  #endif
```
```
// Configuration_adv.h
 #define SDAUTOSTART ENABLED(SDSUPPORT) && DISABLED(ULTRA_LCD)
```
?

p3p, 2019-03-01:
auto0.g will be executed at boot if found.

I'm not sure why the autostart feature is disabled with an LCD, but having a config option for it doesn't seem unreasonable, @thinkyhead may know more about why it is setup that way.

I'm thinking about using the M997 gcode as a reset command, on the lpc1768s it will be a firmware update if a bin is available so it is using it correctly .. mostly

p3p, 2019-03-02:
@Andrei-Pozolotin M997 support is now merged, you can use it to reset after transferring the firmware to the sd card.

Andrei-Pozolotin, 2019-03-02:
1. M997 works great, thank you.
2. please review #13288

closed 2019-03-02 by thinkyhead