MarlinFirmware/Marlin #13280: [FR] (feature request title) G60 is to save the current position and G61 is to go to saved position

lunar711, 2019-03-01:
### Description

### Steps to Reproduce

1. [First Action] G60 (saves current coordinates)
2. [Second Action] (other gcode commands)
3. [Third Action] G61 F1000 (go to saved coordinates at 1000mm per minute)
[image]
 

#### Additional Information
[image]
* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-03-01:
attach 2 config as zip file

lunar711, 2019-03-01:
Uh sorry but i don't understand you (as in which config file)?

boelle, 2019-03-01:
did you read the template when you created the issue?

clearly not since you did not bother edit the title

it's not even clear if this is a question or bug report or a request for a new feature

lunar711, 2019-03-01:
Oh I'm very sorry! I meant this to be a request for a new feature.

Moyster, 2019-03-01:
> clearly not since you did not bother edit the title

No harm meant, but you seem to take this really personally and you probably don't want to bark every time someone does a mistake or simply isn't familiar with how issues "really work" and what they're meant for.  
I think "issues housekeeping" is fine as long as it doesn't come with an attitude.
Also, did you consider "moving every non-bug related issues" to mediums such as discord and facebook groups (both un-indexed on search engines) won't change a thing ? If people actually can't "google their problems", they'll turn to the first place that "makes sense" (ie. github issues).  
I get that you're trying to improve things around, but it looks to me as you don't have a well thought-through approach on dealing with projects and their issues (and issues' issues).  
Certainly not the right place to express my 2 cent, but I felt the need to anyway.

lunar711, 2019-03-01:
Well I did make a glaring mistake so you can't get angry on "boelle"...

boelle, 2019-03-01:
and i might have sounded angry or pissed, but i just keep seeing the same mistake over and over

sadly i do come with an attitude, sometimes it's very little and sometimes it  to much. nobody is complete

about questions... that is not me that invented that discord and facebook are for those places, others made that and i just think its an great idea

thinkyhead, 2019-03-03:
Thanks for the request @lunar711 — I have an E3D Toolchanger and we have added some tool-changing support to Marlin already. However it is not as comprehensive as the support in RepRapFirmware, so we have some catching up to do.

That said, the `sys` and `macros` scripts for the E3D Toolchanger don't use `G60` or even `Tn Rn`. So I guess it's up to slicers to generate the needed G-codes on tool-changes.

thinkyhead, 2019-03-03:
> Also, did you consider "moving every non-bug related issues" to mediums such as discord and facebook groups (both un-indexed on search engines) won't change a thing?

http://forums.reprap.org is also suggested. We don't want to do user support using our issue tracker, and since we've begun being more strict about this it has had a very positive effect on developer collaboration.

lunar711, 2019-03-03:
> Thanks for the request @lunar711 — I have an E3D Toolchanger and we have added some tool-changing support to Marlin already. However it is not as comprehensive as the support in RepRapFirmware, so [...]

Thanks for the reply! I have checked out the E3D tool change code, but what g code initiates it to park and what gcode initiates it to return?

thinkyhead, 2019-03-06:
I looked closer at the `G60` description, and it mentions that RepRapFirmware automatically saves the current position to slot 2 on a tool-change, so that you only need to do the `G0`/`G1` with `R2` for the move back — presumably after a priming and wiping action.

dystechnic, 2019-03-09:
I second this feature request!. 
I use this in some tool changing scripts for a 3-in-1-out switching hot-end but without G60 and G61 the result is somewhat less in comparison to what I was used to. 

This is what I am looking for in regard to functionality.

>**G60**
Save current position coordinates (all axes, for active extruder).S[SLOT] - specifies memory slot # (0-based) to save into (default 0)
**G61**
Apply/restore saved coordinates to the active extruder.
X Y Z E - Value to add at stored coordinates
F[speed] - Set Feedrate
S[SLOT] - specifies memory slot # (0-based) to restore from (default 0)
>

For some reason **G0/G1** with **R*** doesn't work as wel for me, I have a custom build printer with a not so common DUE based controller. Maybe that's why. But G60 and G61 did the trick for me in the past.

lunar711, 2019-03-09:
That's exactly what this request is for (multi-material 3D printing). Basically I wanted to reproduce this here:-

https://www.youtube.com/watch?v=XDZzQ2S6jt4&feature=youtu.be

But on a hardware level (so I can use any slicer, the one 3DMN uses is Simplify3D which is not free).

maxlinux2000, 2019-04-13:
[OT] It would be better to leave Facebook. Why do not you mount a real forum with for example phpbb3 or another platform always opensource?

For example, I have definitively closed all my Facebook and twitter accounts, and like me many people.
So I propose you to create an alternative space to facebook.
If you like the Facebook system you could use OSN (https://www.opensource-socialnetwork.org/)
If you need web space, I can ospitar in my humble server linux.
regards

Hans007a, 2020-01-07:
Here You go for G60/G61 first try:
https://github.com/Hans007a/Marlin/tree/bugfix-2.0.x

maukcc, 2020-05-04:
We tested and works. very nice
1 addition that we would like to see is that the positions are being send to the host (like M114), so the host can do stuff with that info. Or if you send G60 Q0 it will send the stored position of S0.

Simonlebucheron, 2022-08-10:
Is it possible to save G60 coordinates to EPROM ?
And to be sure, does this position is being send to the host, as requested previously ?

ellensp, 2022-08-10:
Eeprom is a bad idea, it only has finite number of writes in its life cycle. You could exceed that in one gcode file.

Simonlebucheron, 2022-08-10:
That is relevant, and what about a register of custom saved position ? Maybe it already exist ?

closed 2026-05-16 by ellensp