MarlinFirmware/Marlin #13285: Dual Zone Heatbeds on the RURAMPS4D board does not work

rusnob, 2019-03-02:
Hello

I want to set up Dual Zone Heatbeds on the RURAMPS4D control board with Marlin 2.0 firmware

I have
1) RURAMPS4D 1.3 control board that supports dual heatbeds.

2) Two heating beds for 220V and two thermocouples.

Task
1) connect the Heatbeds to the board and control the heating of the two zones through the G-code or via the Simplefly 3d interface.

Google did not help, the one that found https://github.com/MarlinFirmware/Marlin/issues/4112

What I changed in marlin 2.0!

Wrote code in Configuration.h

 1. #define TEMP_SENSOR_BED2 1

in pins_RURAMPS4D_13.h

 1. #define HEATER_BED2_PIN 6 // BED H2
 2. #define TEMP_BED2_PIN 0 //

[image]

All firmware files
[Marlin-bugfix-2.0.x-new.zip](https://github.com/MarlinFirmware/Marlin/files/2922150/Marlin-bugfix-2.0.x-new.zip)

Compiled, when entering the command M105 does not show the temperature of the second zone.
 
How to manage the second zone?
What I did not specify in a config?

Thank you, really looking forward to your comment on this.

boelle, 2019-03-02:
marlin does not support 2nd heat bed which is needed to make it work

so you should change title to: [FR] Dual Zone Heatbeds

you dont have to mention Ruramps as this can work on all board as long there is a free control pin and a free input for a thermistor

boelle, 2019-03-02:
also remove the zip file, since this is now a feature request we dont even need the config files

you can in fact remove all text after your picture

boelle, 2019-03-02:
you can replace text before picture with:

Hello

I want to set up Dual Zone Heatbeds on the RURAMPS4D+arduino DUE control board with Marlin 2.0 firmware

Hardware:

RURAMPS4D 1.3
Two heating beds
two thermocouples.

My feature request is  similar to #4112

What I changed in marlin 2.0 so far:

In pins_RURAMPS4D_13.h i have added

#define HEATER_BED2_PIN 6 // BED H2
#define TEMP_BED2_PIN 0 //

Can i control it directly via Gcode?

thinkyhead, 2019-03-03:
We already have multiple heated beds (zones) on our TODO list.

closed 2019-03-03 by thinkyhead