MarlinFirmware/Marlin #13287: [BUG] (short description)

jmdearras, 2019-03-02:
### Description

latest code ignores M190 wait for bed heat goes straight to Hotend heating.

M140 S60
M190 S60
M104 S235 T0
M109 S235 T0

### Steps to Reproduce

1. start print
2. note bed and hotend temps set
3. print starts when hotend is at temp, bed still warming

**Expected behavior:** [What you expect to happen]
should heat bed before heating hotend
**Actual behavior:** [What actually happens]
Both heat at the same time and print starts before bed heated
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

FernandoGarcia, 2019-03-02:
I have experienced the same problem right now with latest Bug fix 2.0.

boelle, 2019-03-02:
@jmdearras remember to edit title to include a description and then attach your 2 config files as a zip file

boelle, 2019-03-02:
but maybe your "issue" is not an issue but your own error, this is how i do start code and it heats the bed first and then nozzle

M851 Z-0.215
G28 Z; Home Z
G28 Y; Home Y
G28 X; Home X
G29
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G92 E0 ; zero the extruded length
G1 X50 Y0 F3000 ; go to the middle of the front
G1 Z0.1 ; close to the bed
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G1 X100 E10 F200 ; make a thick line to prime extruder
G92 E0 ; zero the extruded length
G1 Z0.3 ; lift Z
G4 S2   ;wait 2 sec to bleed off excess pressure

kAdonis, 2019-03-02:
I can confirm, the issue exists!

gloomyandy, 2019-03-02:
What branch of Marlin are you using "latest code" does not really tell us. Also when did you download the branch? Have you been using an older version before that worked? How old was that working version.

@kAdonis same questions for you...

jmdearras, 2019-03-02:
current as of this morning.  Yes, all older versions worked., worked yesterday on 2/24 code.

FernandoGarcia, 2019-03-02:
I have downloaded the code today and it works fine with version of 20/01/2019.

jmdearras, 2019-03-02:
Meanwhile, is there a way with git ( I am not Git literate) to pull the branch at a specific date?  I need to get the printer back on line.

FernandoGarcia, 2019-03-02:
You can work even with this error just heat up the bed before start printing.

To download the code from specific point. 
Browse [commit list](https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x) and select one. So on top right corner select "browse files". Then download the code.

kAdonis, 2019-03-02:
I found the problem, I prepare a PR

jmdearras, 2019-03-02:
> You can work even with this error just heat up the bed before start printing.
> 
> To download the code from specific point.
> Browse [commit list](https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x) and select one. So on top right corner select "browse files". Then download the code.

thank you!  I reverted to 2/24ish and I'm back printing!  The other bug I posted has no workaround I can find.

boelle, 2019-03-02:
i can also find the commit that will pring you back to 2/24

but i assume its 2/24 in us format which in europe ig 24/2-2019

boelle, 2019-03-02:
https://github.com/MarlinFirmware/Marlin/tree/92a4984066bcb0e157d7e1c7c5421de4c1fd684f

that is the last commit on 24/2-2019

closed 2019-03-02 by thinkyhead