MarlinFirmware/Marlin #13292: mks1.4 + tmc2130 spi after the update crazy autohome movement. what can i do?

narsil8, 2019-03-03:
### Description

For your information, I have no experience in coding, and I am an absolute beginner.
Last weekend I upgraded my Prusa-based 3d printer to the mks1.4 board tmc2130 spi version. Before the upgrade, the a4899 step motor driver was in use on the mks gen L 1.0 board. 

As a result of dangling from Google, the step motor on each axis was successful in the host program. 
There is nothing wrong with each motor moving, but once you press Auto Home, the movement of the x-axis is strange and the movement of the z-axis is very strange. 

When the z-axis is moved separately, the step motor is normally operated simultaneously.

Pressing AutoHome on the x-axis moves to the z-axis before AutoHome starts. I don't know why.
And the movement of the z-axis is apparently abnormal. 
The video was filmed behind a 3D printer.

Here are my marlin firmware settings and a YouTube video link. Please help me.

#### Additional Information
my firmware setting marlin bugfix 1.1.x ver.
http://naver.me/GuiO4L6E

autohome movement 
https://youtu.be/__c4sUR-Qtc

z axis movement
https://youtu.be/fr1NGfieYq8

boelle, 2019-03-03:
one thing you can do is using bugfix 2.0 to see if the error has not been fix'ed

next attach your 2 config files as s zip

boelle, 2019-03-06:
@narsil8 tried what i suggested 3 days ago?

narsil8, 2019-03-07:
> @narsil8 tried what i suggested 3 days ago?

It was settled two days ago. I changed all the firmware. So I don't know the exact reason. Thank you.

boelle, 2019-03-07:
@narsil8 maybe click the close button below?
you can always open new issue if you have problems in the future

closed 2019-03-07 by narsil8