MarlinFirmware/Marlin #13300: Babystepping crashes on my delta printer with marlin 2.0.x bugfix

MSpencer0, 2019-03-03:
### Description

Compiling Marlin 2 bugfix for a ANET A4 delta (Melzi board with atmega 128k) and 20x4 LCD controller.

Activated babystepping with doubleclick feature during print as well as
- NO_WORKSPACE_OFFSETS
- MIN_STEPS_PER_SEGMENT set to 2 -> for 20um on a A4

### Steps to Reproduce

1. ) Printing from SD card. First layer started and entering babystepping via doubleclick.
2.) Try to adjust with up or down button
-> Crashes / Hangs up, after approx 2 sec all actuators are shut off. Have to do a power-on-reset to get the printer up and running again.

**Expected behavior:** [

Babystepping should work.

**Actual behavior:** 

Hangs up.

#### Additional Information

Can you reproduce or is it a known issue?

boelle, 2019-03-03:
remember to attach your 2 config files as a zip file

MSpencer0, 2019-03-04:
Configured down with very few compiler switches and even without "doubleclick" but only babystepping accessible via TUNE-menu. That config in zip attached, but problem still exists.

Now migrating config to Marlin 1.1.x release and I will try again.

MSpencer0, 2019-03-04:
I can provide the following summary so far:
Tested with "equivalent" configs  -> provided above for 2.0.x bufgix as well as for 1.1.x release and bugfix version.
Babystepping works both on 1.1.9 release and bugfix.
But it does not work on 2.0.x bugfix as from 26.Feb. 2019.
- Any idea where that feature got lost during migration to 2.0.x?

MSpencer0, 2019-03-05:
Found it! Value overflow in stepper.cpp when calculating EXTRA_CYCLES_BABYSTEP.
This finally has lead to a faulty definition of _PULSE_WAIT which is only(!) used in babystepping.

In old Marlin the default for MINIMUM_STEPPER_PULSE when not given was 2.
That changed in new Marlin to a default value of 1.

This "MINIMUM_STEPPER_PULSE" is taken in stepper.cpp to calculate EXTRA_CYCLES_BABYSTEP
the following way:

  #define EXTRA_CYCLES_BABYSTEP (STEP_PULSE_CYCLES - (CYCLES_EATEN_BABYSTEP))

But it does not check if a higher value is subtracted from a smaller one (which is not the case on a 16MHz CPU with MINIMUM_STEPPER_PULSE of 2 - BUT: It takes place when MINIMUM_STEPPER_PULSE is 1 - like in the new default settings of Marlin 2 )

Solution:
Replace this line in stepper.cpp:
  #define EXTRA_CYCLES_BABYSTEP (STEP_PULSE_CYCLES - (CYCLES_EATEN_BABYSTEP))

With this one:

  #if STEP_PULSE_CYCLES>CYCLES_EATEN_BABYSTEP
	#define EXTRA_CYCLES_BABYSTEP (STEP_PULSE_CYCLES - (CYCLES_EATEN_BABYSTEP))
  #else
	#define EXTRA_CYCLES_BABYSTEP 0  
  #endif

Hope this helps. I would expect this to be an issue on more machines when trying to babystep. Not only on my ANET A4 atmega128 configuration.

tig33r, 2019-03-05:
Is this could be connected with #13309 and #13315?

MSpencer0, 2019-03-05:
Maybe check what happens if you force 
MINIMUM_STEPPER_PULSE to a value of 2 or 3 - See configuration_adv.h 
Is the problem still there then?

Stady234, 2019-04-10:
I just enabled
#define MINIMUM_STEPPER_PULSE 2
on line 1131 of Configuration_adv.h and now my babystep issue looks to be fix.  No more lockup of the machine.

MSpencer0, 2019-04-11:
Hi Stady234,
you are actually describing a workaround for that bug. But in general it is still a bug due to an uncaught exception/overflow.
But for a proper software even a vaild MINIMUM_STEPPER_PULSE of 1 shall not lead to a crash.
Therefore the proposed bugfix would be a clean solution for that.

Stady234, 2019-04-13:
You are correct.. i was simply stating that changing that value back to the previous commit of 2.. solved it.. i have since made the changes in the stepper file as you showed and reset the stepper pulse back to 1 and all is well.

closed 2019-05-17 by thinkyhead

Stady234, 2019-07-24:
Looks like this is still an issue..  I just downloaded the latest commit from today.. 
[image]
Flashed to my delta (Anycubic KLP w/ factory Trigorilla board w/A4988) and left the default "min stepper pulse =1" and sure enough the printer locked up as soon as i turn the lcd knob to any value...

I copied the above outlined replacement lines.. and everything is working as expected.