MarlinFirmware/Marlin #13310: [2.X] Teensy 3.6 onboard SDCARD

TurBoss, 2019-03-04:
Hello!

### Description

Unable to use oboard SD card on a teensy 3.6

### Steps to Reproduce

fat32 formated sd card that cardinfo.ino detects
```
const int chipSelect = BUILTIN_SDCARD; 
```

```
Initializing SD card...Wiring is correct and a card is present.

Card type: SDHC

Volume type is FAT32

Volume size (Kbytes): 15110720
Volume size (Mbytes): 14756

Files found on the card (name, date and size in bytes): 
```

while marling firmware gives 

```
SD init fail
```

**Expected behavior:** 

Working SD CARD using the onboard slot

**Actual behavior:**
```
SD init fail
```

Thanks

boelle, 2019-03-05:
remember to attach your 2 config files as a zip file

and it would be great if you can add [BUG] before [2.X] in title

TurBoss, 2019-03-05:
hello
her I attach the 2 config files
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2932692/Configuration.zip)

shitcreek, 2019-05-15:
@TurBoss  You need to have //#define SDSUPPORT uncommented to enable it; around line 1589 in your config file

closed 2019-05-16 by TurBoss