MarlinFirmware/Marlin #13312: Enable Lin_Advance

ZS1FJK, 2019-03-05:
Hello :)

I have a Lulzbot Taz 6 with Archim2 board (has tmc2130 with trinamic on.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2931987/Configuration.zip)

When I enable Lin advance the extruder motor dose not move.

Step 1 .Enable Lin_Advance
Step 2 . Generate gcade with http://marlinfw.org/tools/lin_advance/k-factor.html
Step 3 . Run g code no Printy Printy.

Expected to print some lines to adjust k factor for Lin_Advance

Dose not extrude. Dose not even prime the nozzle. Also dose not move when k factor on 0

psavva, 2019-03-05:
Try enable Lin advance and see if a simple cube prints

hectori4502, 2019-03-05:
change this: to 1
#define MINIMUM_STEPPER_PULSE 0  //1  Extruder tmc move ok

boelle, 2019-03-06:
@ZS1FJK and please add [BUG] in start of title on your first post

ZS1FJK, 2019-03-06:
@boelle 
have done so. Sorry for that.
@hectori4502 
I am only able to test changes tonight. Will report back as soon as i can.

boelle, 2019-03-06:
it can happen to all of us, but it's more easy for dev people if things start with [BUG]

remember to use bugfix 2.0 when testing

boelle, 2019-03-06:
and of course recheck wiring, check that pins file is correct etc

@marcio-ao i remember that you work for lulzbot, does his config look right?

ZS1FJK, 2019-03-06:
@hectori4502 

Extruder is now moving. 

Thank you

boelle, 2019-03-06:
@ZS1FJK we fixed this?

marcio-ao, 2019-03-06:
> @marcio-ao i remember that you work for lulzbot, does his config look right?

@boelle: I don't know, since I don't have config files to compare to. Due to the number of printer variants we have to generate, our FW generates the configurations pragmatically at compile time and there is nothing to compare against upstream Marlin.

For what it is worth, in our recent "devel" branch we have added unofficial, unsupported options to build for a TAZ 6 with an Archim 2.

boelle, 2019-03-06:
@marcio-ao hehe, but i think @ZS1FJK fixed his problem, does the proposed fix look "reasonble" ?

ZS1FJK, 2019-03-06:
@marcio-ao 
I am unable to get the files to compile from the devel branch. I am unsure how to report these things to you.

marcio-ao, 2019-03-06:
@boelle: Yes, because this workaround has already been [discussed here](https://github.com/MarlinFirmware/Marlin/issues/13040).

marcio-ao, 2019-03-06:
@ZS1FJK: Post a screenshot or log of the error(s) you are getting on the devel branch here. Generally we aren't suppost to support non-stock configs, but I may be able to look it over quickly if you post it here.

boelle, 2019-03-06:
i would not mind he post here, i take that the lulzbot flovour back ports stuff and take stuff from marlin? so why not

hectori4502, 2019-03-06:
I want to share something that I notice when I change from XYZETMC2130 #define MINIMUM_STEPPER_PULSE 0 to 1
although using an XYZTMC2130 E0-LV8729 driver this same effect reduces a lot.
I can only explain it in images. (Someone can explain this effect)
[image]

ZS1FJK, 2019-03-06:
@hectori4502 
I am not sure if they will see this but i will check and see if u can see something.

ZS1FJK, 2019-03-07:
@marcio-ao 
Here is log files and config files.

[Arduino v9 beta Log File.txt](https://github.com/MarlinFirmware/Marlin/files/2941716/Arduino.v9.beta.Log.File.txt)

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2941725/Marlin.zip)
Thank you for your time

marcio-ao, 2019-03-07:
@ZS1FJK: I believe I have fixed those bugs already and you simply need to do a "git pull" in your repo.

marcio-ao, 2019-03-07:
@ZS1FJK: Version 2.0.0.94 had problems compiling under Arduino IDE. I fixed this in 2.0.0.95. The latest you should be able to pull is 2.0.0.97. You can check the version by looking in Conditionals_LulzBot and looking for this line:

```
#define LULZBOT_FW_VERSION ".97" // Change this with each update
```

ZS1FJK, 2019-03-07:
@marcio-ao
Ok will pull the new firmware this weekend and configure it. Thank you

hectori4502, 2019-03-07:
@ZS1FJK 
LIN_ADVANCE_K 0.15

I notice this effect when I change
MINIMUM_STEPPER_PULSE 0 to 1 with TMC2130 drivers (this change I do to operate the extruder)
In the sample of the video you will see that waves are created in the back part.
https://www.youtube.com/watch?v=DCnbbPg7eAc