MarlinFirmware/Marlin #13313: [FR] (Repeat Arc until ...)

kulfuerst, 2019-03-05:
It would be of great help for me if it was possible to have a G2 Arc command repeated either a number of times or favorably until an event is triggered. Specifically, I'd like the print head to move in full circles above the heatbed untill the heatbed has reached a set temperature as part of an end G-code script. I currently cannot find a way to implement this behavior any other way. I imagined the command to look something like this:

G2 I50 J50 R5 (repeat 5 times)
G2 I50 J50 M190 S25 (repeat until bed has reached 25°C)

Thank you very much for your help!

gloomyandy, 2019-03-05:
I think you will need to explain why you need this feature and why it would be of use to other Marlin users.

kulfuerst, 2019-03-05:
Thank you gloomyandy for your reply.
Because I print on an Anycubic Ultrabase I need to wait for the heatbed to cool down before I can remove the print, which can take a substantial amount of time! I have found that my part cooling fan is strong enough to make my heated bed cool down more than twice as fast even at 60% speed (!) than if I let it cool down by itself when I place the print head about 15 cm above it and turn on the fan. I therefore would like to implement this in my end G-Code. However, after an actual print, part of the heatbed is obstructed by the print itself and it is unpredictable where the print head would need to be placed for airflow to reach the heatbed well. Therefore I'd like the print head to move above the print and the print bed in a circular fashion while blasting the part cooling fan to bring my heatbed down to ambient temperature much faster. After the desired temperature is reached, the print head should stop moving and the rest of the end G-code could be initiated. While this may not be useful for everyone, I believe many users printing on glass or on a print surface similar to the ultrabase could safe a lot of time in their workflow if this function was enabled :)

thinkyhead, 2019-03-06:
This can be accomplished by adding a relative Z move and several G2 / G3 moves to your slicer ending G-code. It doesn't need to be added as a core feature of Marlin.

closed 2019-03-06 by thinkyhead

kulfuerst, 2019-03-12:
Hello thinkyhead,
I've been trying around a bunch since you closed the thread, but I cannot get it to work. Depending on the print, the bed takes a different time to cool down. With the relative Z move and a certain number of arcs, the action of the print head is in no way coupled to the temp of the heatbed. Sometimes it continues moving way longer than needed, other times it stops too soon. This doesn't appear to me like an elegant solution, or a solution at all really. Is there any way you might reconsider? Thank you very much for your help.

davidramiro, 2019-03-13:
I agree that this isn't something to be integrated as a core feature, but to whoever this might be useful, I quickly threw together something that does exactly that:

```
   /**
   * M888: Cooldown routine for the Anycubic Ultrabase (EXPERIMENTAL):
   *       This is meant to be placed at the end Gcode of your slicer.
   *       It hovers over the print bed and do circular movements while
   *       running the fan. Works best with custom fan ducts.
   *
   *  T<int>   Target bed temperature (min 15°C), 30°C if not specified
   *  S<int>   Fan speed between 0 and 255, full speed if not specified
   */
  inline void gcode_M888() {
    // don't do this if the machine is not homed
    if (axis_unhomed_error()) return;

    const float cooldown_arc[2] = { 50, 50 };
    const uint8_t cooldown_target = MAX((parser.ushortval('T', 30)), 15);

    // set hotbed temperate to zero
    thermalManager.setTargetBed(0);
    SERIAL_PROTOCOLLNPGM("Ultrabase cooldown started");

    // set fan to speed <S>, if undefined blast at full speed
    uint8_t cooldown_fanspeed = parser.ushortval('S', 255);
    fanSpeeds[0] = MIN(cooldown_fanspeed, 255U);

    // raise z by 2mm and move to X50, Y50
    do_blocking_move_to_z(MIN(current_position[Z_AXIS] + 2, Z_MAX_POS), 5);
    do_blocking_move_to_xy(50, 50, 100);

    while ((thermalManager.degBed() > cooldown_target)) {
      // queue arc movement
      gcode_get_destination();
      plan_arc(destination, cooldown_arc, true);
      SERIAL_PROTOCOLLNPGM("Queued an arc");

      // delay while arc is in progress
      while (planner.movesplanned()) {
        idle();
      }
    idle();
    }
    // the bed should be under <T> now
    fanSpeeds[0]=0;
    do_blocking_move_to_xy(MAX(X_MIN_POS, 10), MIN(Y_MAX_POS, 190), 100);
    BUZZ(100, 659);
    BUZZ(150, 1318);
    SERIAL_PROTOCOLLNPGM("M888 cooldown routine done");
  }
```