MarlinFirmware/Marlin #13314: [BUG] MKS-SBASE/LPC1768 servo vs stepper interference

Andrei-Pozolotin, 2019-03-05:
### Description

servo angle oscillates with stepper motor action

### Steps to Reproduce

1. verify servo is working via `M280 P1 S0`, etc
2. produce some stepper activity, such as `G28`

**Expected behavior:** servo is not affected my stepper actions

**Actual behavior:** servo angle oscillates plus/minus 5 degrees with stepper motor action

the effect is present both when all steppers are connected or disconnected

#### Additional Information

configuration snapshot:
* [Configuration.h](https://github.com/random-builder/Marlin/blob/27cab90154deddc8965da5cce571231daa5e8b69/Marlin/Configuration.h)
* [Configuration_adv.h](https://github.com/random-builder/Marlin/blob/27cab90154deddc8965da5cce571231daa5e8b69/Marlin/Configuration_adv.h)
* [pins_MKS_SBASE.h](https://github.com/random-builder/Marlin/blob/27cab90154deddc8965da5cce571231daa5e8b69/Marlin/src/pins/pins_MKS_SBASE.h)

p3p, 2019-03-05:
That's a lot more deviation than I would expect (5 degrees would need a lot of signal jitter) so I will look into it but it is expected at the moment, Hardware PWM is disabled in the LPC176x framework (time constraints to debug it), although with the pins selected you would not be able to take advantage of it. 

How many concurrent distinct microsecond accuracy signals can you generate in software? The Software PWM can generate up to 20 channels of accurate PWM because they are aligned as long as it has priority, as soon as you want to generate a stepper signal well that needs priority because you really need that to be a consistent signal. The LPC176x is also running an entire USB stack .. people want high frequency PWM for lighting wich would need at least 8 times more interrupts/s, I just "ported" the NeoPixel lib to LPC176x and that needs accurate ns signals per bit and 100's us to update a strip, Marlin has so many features that people expect to work together..

So to sum it up, I'l work on fixing hardware PWM, but even then you only get 6 channels on specific pins that may not be available on your board and all need to run at the same frequency so no lighting if you need servos.

AnHardt, 2019-03-05:
The workaround, to avoid the servo jitter, is to activate `DEACTIVATE_SERVOS_AFTER_MOVE`.

Andrei-Pozolotin, 2019-03-05:
1. @p3p : it seems 6 hardware PWM channels will resolve the problem, thank you.
perhaps default servo mapping in [pins_MKS_SBASE.h](https://github.com/random-builder/Marlin/blob/0961aa022529ebcf38e9a4cbe22054011cb598c7/Marlin/src/pins/pins_MKS_SBASE.h) should be changed or better documented.

2. @AnHardt : `DEACTIVATE_SERVOS_AFTER_MOVE` does not work for me, the servo motor stays energized after the `M280`, see: https://github.com/random-builder/Marlin/commit/0961aa022529ebcf38e9a4cbe22054011cb598c7 ; what am I missing?

gloomyandy, 2019-03-05:
@Andrei-Pozolotin Unfortunately pretty much all of the pins that can be used for hardware PWM are already in use for other purposes. The only pin that is easy to get to and that is not in use is already allocated as SERVO0 (pin P1_23). All of the PWM capable pins are listed in the pins file in the form of a comment. This could do with cleaning up as I suspect it was a copy and paste from the ramps pin file and the pin assignments are not correct for the sbase.

Andrei-Pozolotin, 2019-03-05:
@gloomyandy : this http://smoothieware.org/pwm-capable says:
```
Hardware PWM is on pins 2.0 to 2.5, 1.18, 1.20, 1.21, 1.23, 1.24, 1.26, 3.25 and 3.26
```
which are hardwired on board as follows:
```
// required for core function
 2.0 = X step pin
 2.1 = Y step pin
 2.2 = Z step pin
 2.3 = E step pin
 2.4 = Fan PWM
 2.5 = Bed heater
// more applicable for changes
1.18 = LED1
1.20 = LED2
1.21 = LED3 
1.23 = LED4
1.26 = Y min switch
3.25 = EXP2 / LCD panel
3.26 = EXP2 / LCD panel
```
since I use 
* AUX1/TFT not LCD, 
* have custom cable soldered to LED1, LED2, LED3, LED4, 
* can re-configure `Y min switch` to use `Y max switch`,
I should be able to choose from these 7 h/w pwm:
```
P1_18, P1_20, P1_21, P1_23, P1_26, P3_25, P3_26
```

gloomyandy, 2019-03-06:
You have missed 1.24 which is X Min. Also are you sure you have the various LED pins assigned correctly. My understanding is that on the SBase they are:
LED_PIN            P1_18
LED2_PIN           P1_19
LED3_PIN           P1_20
LED4_PIN           P1_21
and that pin P1_23 is only attached to the header J8_3

As I said above very few of the PWM pins are easily available for other purposes. I'm not sure how well using the LED pins will work (I think at least one of them 1.18 is toggled on and off by the watchdog  and I'm not sure how having the LED connected will impact the use for things like a servo). The two endstop pins have additional hardware (resistor/capacitor filter), which again may impact the use of these pins for some purposes.

You also need to be careful because there are only 6 PWM channels and these can only be assigned to certain pins. So for instance pwm channel 2 can be used with pins 3.25, 2.1, 1.20, similarly channel 3 is associated with pins 3.26 2.2, 1.21 so you can only use one pin from each set. 

So my understanding of the PWM capable pins is
```
 2.0 = X step pin [chan 1]
 2.1 = Y step pin [chan 2]
 2.2 = Z step pin [chan 3]
 2.3 = E step pin [chan 4]
 2.4 = Fan PWM [chan 5]
 2.5 = Bed heater [chan 6]
// more applicable for changes
1.18 = LED1 [chan 1]
1.20 = LED3 [chan 2]
1.21 = LED4 [chan 3]
1.23 = J8_3  [chan 4]
1.24 = X Min Switch [chan 5]
1.26 = Y min switch [chan 6]
3.25 = EXP2 / LCD panel [chan 2]
3.26 = EXP2 / LCD panel [chan 3]
```
@p3p do you know if any of the PWM channels are used for things like fan control or heater control? I can't easily search the source at the moment.

Andrei-Pozolotin, 2019-03-06:
@gloomyandy RE: `Also are you sure you have pins assigned correctly`
No, at some point I have to take the board out and trace all these pins to the mcu with a multimeter.

p3p, 2019-03-06:
@gloomyandy I'm pretty sure Marlin generates it own signal for heaters and fans.

> @AnHardt : DEACTIVATE_SERVOS_AFTER_MOVE does not work for me, the servo motor stays energized after the M280, see: random-builder@0961aa0 ; what am I missing?

I'm not sure why that doesn't work for you, I just checked here and the PWM signal definitely stops after `SERVO_DELAY`

gloomyandy, 2019-03-06:
My version is based on the schematics that MKS released for a slightly earlier version of the board. You can find them here: https://github.com/pixel3design-hub/MKS-SBASE-FULL-DOCUMENTED Some things may have changed but so far most things seem to match up.

Andrei-Pozolotin, 2019-03-08:
@gloomyandy 

1. I verified traces on the SBase V1.3 board, this is still correct pinout:
https://github.com/pixel3design-hub/MKS-SBASE-FULL-DOCUMENTED/blob/master/MKS%20SBASE%20SCH_V12.pdf
```
LED_PIN  P1_18
LED2_PIN P1_19
LED3_PIN P1_20
LED4_PIN P1_21
```

2. conveniently, `EXP1` has 3 `NC` pins, lending itself for "custom cable" 
in the form of awg-30 / 0.25 mm transformer wires running from LEDs to the EXP1

[image]
[image]

gloomyandy, 2019-03-09:
What are you planning on using those connections (from the LEDs) to EXP1 for? Do they work? Some displays may have them connected so take care if using a standard cable LCD!

Andrei-Pozolotin, 2019-03-09:
1. plan to use for enclosure heater control. 2. seem to work fine so far. 3 no LCD, using AUX/TFT.

gloomyandy, 2019-03-09:
Sounds good! Nice wiring, did you have to drill a hole to get the wiring through or was there one there ready to be used?

Andrei-Pozolotin, 2019-03-10:
The hole is part of RJ-45 socket mount. There are 4 like this: 2 from USB and 2 form RJ-45.
The secret technique is: melt the solder in the hole, then stick a toothpick. :-)

boelle, 2019-03-13:
@Andrei-Pozolotin is this solved?

p3p, 2019-03-13:
@boelle It's not solved, but its not really an issue with Marlin either, I'm fighting with the hardware PWM trying to get it functional enough to include in the framework .. problem is putting the peripheral into latched mode to avoid any inconsistency on the signal makes it randomly miss a change in duty cycle completely, I'm unsure if its my stupidity, a compiler optimisation problem or undocumented feature,

boelle, 2019-03-13:
@Andrei-Pozolotin i was just unsure if what @Andrei-Pozolotin did was a fix or trick :-D

p3p, 2019-03-26:
@Andrei-Pozolotin if you update the platformio lpc176x framework the hardware channels are enabled again, how long they stay enabled depends on bug reports ;)

Andrei-Pozolotin, 2019-03-29:
@p3p Chris:
1. thank you for your efforts, can you please highlight where/what have you changed?
2. I verified with latest pull / snapshot (below) that there is no change in servo vs stepper interference,
i.e. there is still same amount of servo shaft vibration when steppers are performing homing operation
https://github.com/random-builder/Marlin/tree/483abfc16963470c3b61c92079c7f08c68d60227

p3p, 2019-03-29:
Make sure you have updated to framework-arduino-lpc176x @ 0.1.2 and are using a hardware pin. (https://github.com/p3p/pio-framework-arduino-lpc176x for compatible pins)

You can see the commits since last release in the [releases](https://github.com/p3p/pio-framework-arduino-lpc176x/releases). I just refactored and re-enabled the PWM hardware, hoping I had worked around the bugs with pulse width latching that we had.

Andrei-Pozolotin, 2019-03-30:
1. Success: I confirm that upon upgrade to latest framework-arduino-lpc176x vibration is gone. 
2. Config snapshot:
https://github.com/random-builder/Marlin/tree/5916a0895eff3711633af4cede8c094c585989fc
3. Thank you!

closed 2019-05-17 by thinkyhead