MarlinFirmware/Marlin #13316: [FR] Trinamic smooth acceleration over stealthChop/spreadCycle hybrid threshold

VanessaE, 2019-03-05:
### Description

So...  I have a BigTreeTech/BIQU SKR v1.1, and am able to use TMC2208's in single-wire UART mode, thanks to work associated with https://github.com/MarlinFirmware/Marlin/issues/13197 others (I'm not sure just who to actually thank. @gloomyandy ? :smiley: )

There is one thing that I think should be addressed, but it's not specific to the 2208.  Rather, it applies to  all Trinamic drivers, I guess:

If a move will cross the stealthChop/spreadCycle hybrid threshold while accelerating and/or decelerating, some way to avoid confusing the TMC's math is kinda needed, as they get noisy when moving at speeds close to the threshold.

From what I've read, this may be because Marlin switches to and from double- or quad-stepping during the accel/decel phase.

It's my understanding that Trinamic drivers can move the motors on their own, with commands sent over UART, so could it be possible to split the accel/decel periods into many short moves and let the driver handle them, instead of banging on the step/dir lines?

### Steps to Reproduce

1. Set a fairly low hybrid threshold, say 150 mm/s
2. execute a long travel move at, say, 300 mm/s with around 1000 mm/s² acceleration -- the move must be fast enough to invoke double- or quad-stepping mode on the selected axis.
3. listen for a "thunk" or "chattering" sound during accel/decel.

The motor's not skipping/stalling, it's just noisy.

p3p, 2019-03-05:
The noise can't have anything to do with double stepping if you are using the configs you posted in the other thread, the LPC176x boards do not start doing that until 80K steps/s, you use 80 steps a unit @ 500mm/s so 40K steps/s.

VanessaE, 2019-03-05:
Well so much for that theory :smiley:  ...  still needs addressed though.

boelle, 2019-03-06:
running 12V or 24V ?

teemuatlut, 2019-03-06:
> It's my understanding that Trinamic drivers can move the motors on their own, with commands sent over UART, so could it be possible to split the accel/decel periods into many short moves and let the driver handle them, instead of banging on the step/dir lines?

Only the TMC5xxx drivers (of the ones we're supporting) have the ramp generator you're talking about. This is how the Trinamic TRAMS platform worked as it had no step and dir lines into the drivers and I did have an updated firmware but that's already a year old at this point.

> If a move will cross the stealthChop/spreadCycle hybrid threshold while accelerating and/or decelerating, some way to avoid confusing the TMC's math is kinda needed, as they get noisy when moving at speeds close to the threshold.

The switching velocity is not the same when accelerating and decelerating, but rather they have a hysteresis associated with it. We're already using the larger setting (also the default) with TMC2130 and TMC2208 does not have it configurable.
[image]
I think the image should say `TPWMTHRS + 1/16`.

There's this excerpt from the datasheet that may hint at what you're having.
> (6.5) A jerk occurs when switching at higher velocities, because the back-EMF of the motor (which rises with the velocity) causes a phase shift of up to 90° between motor voltage and motor current. So when switching at higher velocities between voltage PWM and current PWM mode, this jerk will occur with increased intensity. A high jerk may even produce a temporary overcurrent condition (depending on the motor coil resistance). At low velocities (e.g. 1 to a few 10 RPM), it can be completely neglected for most motors. Therefore, consider the switching jerk when choosing TPWMTHRS. Set TPWMTHRS zero if you want to work with stealthChop only.

Personal opinion, use low hybrid threshold with the updated chopper parameters. At this point the spreadCycle is almost as quiet as stealthChop.

VanessaE, 2019-03-06:
@boelle Mine's a 12v system.

@teemuatlut,

>Only the TMC5xxx drivers (of the ones we're supporting) have the ramp generator you're talking about

nono, I was suggesting *Marlin* generate the accel/decel curves more or less as is done now, but send that to the driver as a chain of very short simple moves, gradually increasing or decreasing their speed, letting the 2208's internal pulse generator turn the motors, as described by section 12 of the datasheet (page 61), instead of doing it all through step/dir.

So from my point of view, a 200 mm travel, as commanded by G-code, would become maybe a few hundred move operations, instead of tens of thousands of bit-banged pulses.

All of that said, I can't make any sense of the datasheet regarding how such operations are issued.  In any case, would this even help?

> use low hybrid threshold with the updated chopper parameters

Mine's still at the upstream default of `CHOPPER_DEFAULT_12V`, whatever that equates to.  I'm not aware of these "updated" parameters, and as it pertains to Marlin, I have found nothing that adequately explains why I would want to change this setting, let alone use the custom option (voltage aside, since that's obvious).  I'm at commit 2513f6b5, if that tells you anything. :smiley:

boelle, 2019-03-06:
go for 24V, i had the same issue and Trinamic told me that noise starts to drop at about 18V so 24V was the recommended, just thoughts

VanessaE, 2019-03-06:
@boelle yeah, I've seen similar comments elsewhere, but that's money I can't spend right now.

boelle, 2019-03-06:
same here, i ended up finding a 12V->24V stepup converter and feed the drivers directly

boelle, 2019-06-22:
missing label

VanessaE, 2019-09-19:
I was giving this some thought lately...  how about just having Marlin force the driver to spreadCycle mode when the planner encounters a travel move if the `F` move speed would exceed the configured Hybrid threshold, do the move entirely normally, and switch back to stealthChop mode for the next print move?

In other words, anticipate the Hybrid switchover and just do it ahead of time.  No complicated acceleration emulation or relying on features only some chips have.

LemaruX, 2019-09-26:
Would be nice to have a resolution to this. I have now disabled the hybrid threshold and leave it in stealthchop as the noise isn't great and loads louder than my old A4988s

cnf, 2020-04-24:
Came here for the same reason, @VanessaE suggestion sounds nice? the sound when it switches from stealthchop to spread cycle mid movement is quite jarring :P

VanessaE, 2020-06-08:
*bump* 

Is there just no interest in this?

VanessaE, 2025-06-19:
Just poking my head in for a moment...  I was doing some speed tests these past couple of days and re-discovered this issue, I'd totally forgotten that I filed it!  🤣

It does still need to be addressed, but in the meantime, I've taken to simply locking X and Y into spreadCycle mode when printing with my higher-speed slicer presets, by way of gcode embedded therein (that way I don't have to remember to change modes).

The difference it's made in noise is surprising.  While my machine's main sources of noise seem to be from its X carriage (I'm no engineer 😛 ), a big chunk of the noise was coming from the hybrid switching, just as in the past.  I just tended to ignore it because only travel moves tended to be fast enough to trigger the switching, but now with higher-speed print presets, the switching happens way more often, in print moves as well as travels.  That ain't good.

Staying in spreadCycle mode is WAY quieter than allowing the drivers to switch modes mid-move.

rthresher, 2025-06-24:
Why not servo?