MarlinFirmware/Marlin #13318: [FR] automatic case light

eikeime, 2019-03-06:
### Description

automatic turn on case light 
### Steps to Reproduce

1, setup 2 pwm value for case light, one for idle the other for running.
2, exactly like controller cooling fan, any action trigger case light to running mode, after idle for sometime like 60s turn it to idle mode.
3, if it's possible, lcd menu entry for idle pwm value settings

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-03-06:
maybe add gcode in slicer start gcode that turns it on? that would be 99.99% autoamtic :-D

eikeime, 2019-03-06:
> maybe add gcode in slicer start gcode that turns it on? that would be 99.99% autoamtic :-D

Yes that would work, I setup octoprint to do thie automaticly, but that still cant solve homing without light problem, no light nothing would show on webcam.

boelle, 2019-03-06:
so you  control via octoprint and the only problem is no light when homing from "control"

have you looked if there is not a plugin that can add a light button?

eikeime, 2019-03-06:
yes there is "custom control editor", i put two button for that, but it's manually, 
Octoprint can't do turnoff light after delay a certain time, i tried "G4 S60 M355 S0" this works, but during the delayed period the printer is inoperable, there is a "PSU Control" plugin could solve this, but it's already used for psu control.
I tried to parallel connect case light with controller fan, fan pwm value is too bright, T_T 
Also i really don't want to rely on octoprint for printer hardware automation.

boelle, 2019-03-06:
i agree, i dont have case light YET but i will have at some point

hmm marlin also have psu control if it's an ATX psu, maybe that could be reused to have the case light come on automatic?

but then there is also neopixel led control and i forgot the description on how that works, but its some thing that you can have one color when idle, and another color when bed is heating and then another when nozzle is heating and another when print starts

@thinkyhead thoughts?

eikeime, 2019-03-06:
yes, different color is a great idea, although it's a little bit like rgb trend on pc, at least on 3d printer it's not just for show off.

boelle, 2019-03-06:
well its a bit of showoff as most have either a LCD or Octoprint and can get the same info that the RGB led's provide

but if you only have webcam picture its a great idea

7eggert, 2019-03-08:
A similar but more advanced feature is already implemented, but many of the colors are pre-set.

- Power-on color ist menu-adjustable or compiled in;
- heating color fade,
- white printing light and
- green finished light is hardcoded and you have to press a button after printing;
- then the light turns off.

(I'm annoyed by the "press to resume" message and just deleted the code on my compile)

boelle, 2019-03-08:
and what if light is off and he homes the machine?

7eggert, 2019-03-09:
There is no change on homing; but on heating the light will turn on (Color fade). (Looking at 1.1.9)

boelle, 2019-03-09:
and what @eikeime wants is that the case light comes on when homing :-D

7eggert, 2019-03-09:
Cherry-pick this ad-hoc hack (or use the patch I attached), it will just set the default colors on homing:
github.com/MarlinFirmware/Marlin/commit/2d8092adb7be4a2e4eb15310c0ca519546cd7e2d

I think more discussion is needed on a decent LED feature.
- user-defined status LED
- Status LED vs. colors set by gcode
- timeouts
- Support for more sets of LED + gcode extension
- support for mixed drivers (PWM + Neopixel + …)
- ???

[foo.patch.gz](https://github.com/MarlinFirmware/Marlin/files/2948581/foo.patch.gz)

7eggert, 2019-03-10:
For 2.0:
github.com/MarlinFirmware/Marlin/commit/d073beb2737f51f31b851e401c2cd1c7b13982cf

boelle, 2019-06-22:
missing label

Fixx, 2019-10-09:
A feature will be great like: case light timeout after specific time value, and a trigger like encoder click or rotation brings case light back on.

senseisimple, 2020-10-21:
Sorry for my Commit/PR mess!
#19831 adds simple functionality to run gcode as part of the power on/off routine where PSU_CONTROL is enabled (whether it's manual M80/M81, auto on, or timeout).

This includes turning the case light on or off (M355) as part of a power action as well as other commands like G28 or set messages to display M117.

The M355 gcode of course would need to be enabled (and the case light configured) to run, and can be used independently of setting it in the power actions.

closed 2020-10-23 by thinkyhead