MarlinFirmware/Marlin #13319: [FR] (feature request title)

Miran81, 2019-03-06:
Hi . I'm using ramps-fd v1 and 2004 lcd . I downloaded marlin 2.0 bugfix and uploaded in arduino due . I used #define REPRAP_DISCOUNT_SMART_CONTROLLER in configuration.h but its not working. Just squares on screen.2004 lcd or any lcd for u have to use lcd adapter for ramps v1 or v2. I'm using adapter which have 74HC07A module. Please help me.. Thank u

boelle, 2019-03-06:
attach 2 config files as a zip file

boelle, 2019-03-06:
@Miran81 also add [BUG] in start of title of post so it becomes [BUG] Rams-Fd 2004 LCD Problem

Miran81, 2019-03-06:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2935547/Configuration.zip)

rafaljot, 2019-03-10:
is it your config? I can't see any changes.
it is:
`#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB
#endif`
should be:
`#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_FD_V2
#endif`

or _V1 if you have RAMPS-FD v1 

I have RAMPS-FD+DUE+2004. It works for me.

Miran81, 2019-03-10:
@rafaljot ı'm using in code board_ramps_fd_v1 but its not working . Can u share lcd folder and configuration.h ? Its working but #has_caharacter_lcd and some initializing code in lcd()

Miran81, 2019-03-10:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2949306/Configuration.zip)
@rafaljot

rafaljot, 2019-03-10:
Why do you think it is a bug in firmware? 
Are you sure that cables, LCD, etc are ok? Does this configuration works with other firmware or sth?

Miran81, 2019-03-10:
@rafaljot what is  your marlin version

rafaljot, 2019-03-10:
Do we talk about Bugfix-2.0.x ?
but I'm using Arduino 1.9 . When I use PlatformIO I have the same issue.

Miran81, 2019-03-10:
@rafaljot yes ı'm using platformio. I will try ardunio 1.9 but is this beta and arduino web site has 1.8.8

rafaljot, 2019-03-10:
Yes. 1.9 is a beta. 
Unfortunately Arduino 1.9 can't compile Marlin for DUE on OSX :D, so for this project I use Windows (Probably there is solution for this issue)

Miran81, 2019-03-10:
Okey thank u I'm trying 👍  @rafaljot

Miran81, 2019-03-10:
@rafaljot I'm solved :D how ? I'm using arduino 1.8.8 and have some errors for due. 
1) upgrade board for due in boards menu
2)open platformio.ini in vscode and change 'env_default = DUE' 
3)#define MOTHERBOARD BOARD_RAMPS_FD_V1 and #define REPRAP_DISCOUNT_SMART_CONTROLLER   THAT'S ALL . Thank u all 👍

Miran81, 2019-03-10:
@boelle @rafaljot  when cut energy from usb and when I give energy again lcd screen shows squares but when I'm reseting in board everything is fine normal ? why it wants reset ?

boelle, 2019-03-13:
@Miran81 sometimes a reset is just needed

are your problem solved?

Miran81, 2019-03-13:
@boelle when ı cut off energy and plug it in , it wants reset . But when ı'm reset everything is ok.

boelle, 2019-03-13:
yes a reset is needed after plugin, that is normal

so it seems the problem is solved, you can click the close button below

closed 2019-03-13 by Miran81