MarlinFirmware/Marlin #13324: safe homing and bed levelling issue

marquismatt, 2019-03-07:
Hi Guys 

I'm stuck on two things, I've set up ABL in config.h etc in 1.1.9
the first issue is that it wont compensate for the bed unevenness (levelled bed manually beforehand)
and now it wont safe home properly with G28 it will home the X and Y but when it goes to home the Z it hits the right hand side instead of stopping at the centre of the bed. any ideas on what I have missed or have done wrong would be greatly appreciated : ) 

hope it makes sense   

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2940886/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2940887/Configuration_adv.zip)

boelle, 2019-03-07:
have you tried on bugfix 2.0 also? its both 8 bit and 32 bit

marquismatt, 2019-03-07:
I’ll have to have a look at 2.0
over the weekend 

Cheers for the reply

boelle, 2019-03-13:
what result did you get in weekend testing 2.0?

shitcreek, 2019-05-15:
@marquismatt  - any update?

marquismatt, 2019-05-25:
Sorry guys for taking so long to reply. I ended up using 1.1.9 again after double checking everything it worked. I couldn’t find what I missed tho. Thanks for checking up

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

marquismatt, 2019-06-22:
Oops sorry

closed 2019-06-22 by marquismatt