MarlinFirmware/Marlin #13325: [FR] Should Marlin 1.1.9 be moved to it's own Repository?

dcwalmsley, 2019-03-07:
Since DEVs stopped developing Marlin 1.1.9, there are still folks posting issues in this repository along with issues related with 2.x.

Maybe it would be wise to break the old version from this group and move it to it's own repository and so with the issues.  This way you can concentrate on issues solely related to 2.x development.

dcwalmsley, 2019-03-07:
To be clear...   Maybe have MarlinFirmware / Marlin 1.1.9 (8bit) and one for MarlinFirmware / Marlin 2.x (8bit/32bit).

boelle, 2019-03-07:
we tend to ask people to upgrade to 2.0 instead since it has teh same features plus its updated and fixed

personal i wonder why people have not figured that 2.0 is there and its both 32 bit and 8 bit

during my first cleanup frenzy i got the gist that people think 2.0 is 32 bit only and they cant use it

dcwalmsley, 2019-03-07:
So last year, I heard 2.x was for 32bit boards and like many others, felt 
that the 8bit boards were no longer supported.

I came away with the thought that industry was shifting to more robust 
processors with additional storage capacity and that Marlin 2.x was to 
evolve towards that end.  So it doesn't surprise me that many may not have 
realized that 2.x supports 8 & 32 bit boards albeit 8bit boards have 
limitations on both the processor power and overhead storage that some 
newer additions to Marlin 2.x may not work on them.

dcwalmsley, 2019-03-07:
Another question is to wonder why you need 1.1.9 up on the site?  I have 
2.x running on my Azteeg X3 Pro with TMC2130 installed as well as my 
HyperCube with the Einsy Rambo board and all is good.  I know at some point 
hardware upgrade to 32bit may be needed but for now I am ok with Marlin 
bugfix 2.x.

boelle, 2019-03-07:
2.x is both 8 bit and 32 bit :-D

boelle, 2019-03-07:
yeah, you can wonder why to have 2 versions, when 2.x supports both 32 bit and 8 bit

maybe @thinkyhead can add his thoughts?

boelle, 2019-03-07:
those come in to trouble with 2.x and 8 bit are the "speedfreaks" ie those that want to go fast without thinking if their hardware can even cope, i have been in that trap too and the result is either jerky motion or the board will reset itself

just having a display eats a lot of cpu cycles...

dcwalmsley, 2019-03-07:
Good point.  I haven't running into performance issues as I stay below 
60mm/min.

thinkyhead, 2019-03-07:
There is only one current release, and that is 1.1.9. Marlin 2.0 has not yet been released, is under constant development — and is in pre-alpha state for most 32-bit platforms. There are points where the development branch is pretty bug-free, and it has been recommended to try the development branch if a bug in 1.1.9 is getting in the way.

We did try a 'dev' separate repository, but it didn't change things from our side. It meant tracking two sets of issues instead of one, and sharing commits and links between the repositories, additional traffic direction, permission maintenance, and so on, only added to the workload without bringing any special benefit.

Right now we do want to know about bugs and feature requests, whether they relate to the current release or to ongoing development, since issues often pertain to one another.

When we tag and release Marlin 2.0 it will be at the top of the same Releases page where 1.1.9 now resides, and it will be clear from the release title that it is for both 8-bit AVR and 32-bit ARM, and the successor to 1.1.9.

We have good development momentum on all fronts (HALs, LCDs, UI, features, bugs) at this time, and reasonably good processes that allows PRs to be discussed, cleaned up, improved, and matured before they are merged. The only thing that gets PRs cleared is my having the time to fully absorb the changes and workshop them for the hours or days required to bring them into shape.

And so I am now debugging power-loss recovery, and then later I'm working on the configurator and configuration documentation for the website. Never a dull moment…

closed 2019-03-07 by thinkyhead

dcwalmsley, 2019-03-08:
I for one sincerely appreciate all the effort you and other Dev's have made 
to Marlin.

Thank you for all you good folks do.