MarlinFirmware/Marlin #13326: [Bugfix 2.0.x] mks 1.4/tmc2208 no motion on any axis - steppers communicating via UART

freighttt, 2019-03-07:
### Description

Have tmc2208's running in uart, but all steppers are powerless. the drivers communicate just fine. rolled back to 1.1.9 and i have motion. Using the most curren 2.0.x and havent messed with anything outside of standard setup. running an ender 3 with mks 1.4, so the board doesnt have PSU control and as such the variable in marlin for psu control is 0. when i ask it to home it does nothing and then eventually says homing failed, please reset. very very confused.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2942445/Configuration.zip)

p3p, 2019-03-07:
```
  /**
   * Software enable
   *
   * Use for drivers that do not use a dedicated enable pin, but rather handle the same
   * function through a communication line such as SPI or UART.
   */
  #define SOFTWARE_DRIVER_ENABLE
```

Have you physically configured your drivers to work with this option? I'm not a TMC guy but this [comment](https://github.com/MarlinFirmware/Marlin/issues/13197#issuecomment-469583276) seems to say you need to change the hardware.

boelle, 2019-03-07:
maybe @teemuatlut knows?

freighttt, 2019-03-07:
@p3p yeah, they’re wired for uart and connected as such. M122 responded with OK connections across all 4 drivers.

p3p, 2019-03-07:
The drivers need modified as mentioned in the comment I linked to specifically to allow the `SOFTWARE_DRIVER_ENABLE` feature to work, this is extra to be wired for UART mode.

freighttt, 2019-03-07:
@p3p I’ll have to double check my config when I get home, I’m pretty sure it’s enabled but I’ll feel awful stupid if it isn’t haha

p3p, 2019-03-07:
The problem is that it is enabled, not that it isn't, ^^ the `SOFTWARE_DRIVER_ENABLE` feature is relatively new and not available in 1.1.9, so as your drivers work with 1.1.9 when you test I'm pretty sure you're drivers are not wired up for it.

gloomyandy, 2019-03-07:
If you are using that option then you need to ensure that the drivers are actually enabled by default. This means that the enable pin on the driver must go to ground. This may mean you have to remove the pin that will normally go to the enable line on the module and connect a wire from the driver enable pad to ground. Note that if you don't do the hardware mod you can not rely on the pin normally used for enable to do the job because when you define SOFTWARE_DRIVER_ENABLE that pin will not be set to output and will typically be "floating".

The obvious question is why have you enabled this feature? Do you really need it? It has been used on boards that have limited pins available to free up the cpu pins that are normally used for driver enable to be used for other purposes (like chip select when using SPI, or perhaps as a UART line). But really if you don't need the extra pins you are better off not using SOFTWARE_DRIVER_ENABLE.

freighttt, 2019-03-07:
@gloomyandy I will double check my file when I’m home, I think I follow what your saying, but this is all a bit out of my realm of expertise so bear with me :)

teemuatlut, 2019-03-08:
`SOFTWARE_DRIVER_ENABLE` isn't a feature you enable to gain something, but rather a necessary option to accommodate certain hardware. Currently it is only needed for Panucatt BSD2660 drivers, but of course there may be others that I'm not yet aware of.

gloomyandy, 2019-03-08:
@teemuatlut it can be useful on boards like the skr v1.1 that have very few pins available. On that board you can modify the driver board to hardware the enable pin and then use this feature to free up the enable pin for use as chip select (or as a UART line). But yes I would agree that it is not something for general use.

freighttt, 2019-03-09:
That was the issue, sorry for the trouble gentleman. Closing this. Thank you!!!

closed 2019-03-09 by freighttt

VIXIVIXIV, 2020-06-14:
@gloomyandy Hopefully commenting on this closed issue isn't impolite, but I figured this would be the easiest way to reach you.

Anyhow, have you been able to get SOFTWARE_DRIVER_ENABLE working on stepper drivers that use UART (of course, after tying the EN pin to ground)? I'm planning on a build using TMC2209s, but I don't want to solder the EN pin ahead of time only to discover that it doesn't actually work that way. @freighttt seems to suggest that is the case, but the response was a bit vague as to whether the solution was commenting out SOFTWARE_DRIVER_ENABLE or adding the soldering required to make it work.

And before you ask, yes, this is (probably) necessary; the stock board doesn't really have pins to spare, so if I can free up the EN pins from the processor, that'll make UART possible without sacrificing other functionality.