MarlinFirmware/Marlin #13327: [FR] Userfeedback with M0/M1 and ExtUI.

coldtobi, 2019-03-07:
### Description

During work on  #13253 (but with using the ExtUI interface) I'd like to implement M0/M1 support for my display, as this is a feature I personally often use in my gcode scripts. The idea is to show an popup on the display with the optional message in M0/M1 and ask the user to acknowledge the popup.

Currently this is not supported by ExtUI... However, I'm not sure to implement it in a way that could be be merged into the Marlin project. One straight forward way would be to directly call out in M0_M1() to an (new) eventhandler, e.g ExtUI::OnUserConfirmRequired(char *msg) resp. an exit handler to stop displaying the pop up (after timeout) is called in M0_M1() directly. Would that be acceptable or what other way would be?

marcio-ao, 2019-03-08:
@coldtobi: Using a new event handler would be how I would approach it.

For the timeout, perhaps if ExtUI::onUserConfirmRequired(char *msg) is called with NULL, it means to hide any popup that was already issued?

marcio-ao, 2019-03-08:
One thing that isn't clear to me is how the ExtUI is supposed to notify Marlin that the user has acknowledged the pop up? Would the ExtUI have to call a GCODE similar to M876 when the user pushed the button? I'm going to ping @InsanityAutomation in this thread since he has being doing a lot of work related to the interaction between Marlin and the host. The interaction between Marlin and the ExtUI probably has similar challenges and the solutions probably ought to resemble each other.

InsanityAutomation, 2019-03-09:
I guess given the documentation on the reprap wiki states heaters are to be turned off following M0 or M1, where the marlin wiki does not, the question is what the intended use of them is with the external display that isn't available elsewhere. An M125 (or m25 if SD printing and advanced pause is off) may be more appropriate if a cancel is not the intention. Regardless, I would think the host prompt output could be reused in a message set and then the handlers called directly instead of using a second buffer internally.

coldtobi, 2019-03-09:
> @coldtobi: Using a new event handler would be how I would approach it.
> 
> For the timeout, perhaps if ExtUI::onUserConfirmRequired(char *msg) is called with NULL, it means to hide any popup that was already issued?

@marcio-ao, sounds good to me, I'll create a PR for better discussion.

> One thing that isn't clear to me is how the ExtUI is supposed to notify Marlin that the user has acknowledged the pop up? Would the ExtUI have to call a GCODE similar to M876 when the user pushed [...]

ultralcd code just reverts wait_for_user to false, but I think the best thing would be to hide the implementation details behind an ExtUI-API, something, maybe void ExtUI::setUserConfirmed(void)?

>I guess given the documentation on the reprap wiki states heaters are to be turned off following M0 or M1, where the marlin wiki does not, the question is what the intended use of them is with the external display that isn't available elsewhere. 

My usage is in a Z-axis leveling script, and it would be unfortunate if it'd turn off the motors; so I like the Marlin way ;) When there is no external display, with HOST_PROMPT_SUPPORT the host can ask to resume processing (M108, see below)

Looking at [the NIST reference implementation documentation](https://www.nist.gov/publications/nist-rs274ngc-interpreter-version-3?pub_id=823374) I think the Marlin's behaviour is correct, quoting from the paper: "a program is stopped by an M0, M1, or M60, pressing the cycle start button will restart the
program at the following line."

>An M125 (or m25 if SD printing and advanced pause is off) may be more appropriate if a cancel is not the intention. Regardless, I would think the host prompt output could be reused in a message set and then the handlers called directly instead of using a second buffer internally.

The M0/M1 docu on Marlin wiki says that "With EMERGENCY_PARSER enabled the M108 command can be used to continue.", so I'm not sure if the side effects of M25/125 are good for this case... 
But, on a second thought, the gcode parser is paused until M0/M1 is acknowleged, so the only commands have an effect will anyway be ones using the emergency parser. So I think flipping the boolean is not wrong?

coldtobi, 2019-03-10:
Has been merged.

closed 2019-03-10 by coldtobi