MarlinFirmware/Marlin #13329: can not compile marlin 2.0 with mks sbase fails neopixel without even having neopixel uncommitted

rathorp80, 2019-03-08:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

p3p, 2019-03-08:
Update PlatformIO frameworks and libraries

J-charles-C, 2019-03-08:
I,

P3P's answer yesterday when I asked the same question:

`Depends how you use PlatformIO to build Marlin.

Terminal:
platformio update or pio update

Atom:
Click the PlatformIO menu then select Update platforms, packages and libraries

Visual Studio Code:
Click the Platform IO symbol (Alien head) on the left menu bar, in Quick access Menu click on Updates -> Update All

`

best

boelle, 2019-03-13:
@rathorp80 did updating PlatformIO frameworks and libraries solve the issue?

closed 2019-03-15 by p3p