MarlinFirmware/Marlin #13332: Marlin 2.o Wont compile

cinealfa, 2019-03-08:
### Description

Latest Bug 2.o Branch has error in compiling.

### Steps to Reproduce

1. Fresh install of Marlin source
2. Using archive of known good config files dated 1/31/19
3. initiate compile

**Expected behavior:** [What you expect to happen]
Program compiles
**Actual behavior:** [What actually happens]
passes sanity check then throws the following error:

In file included from Marlin/src/lcd/dogm/status_screen_DOGM.cpp:32:0:
Marlin/src/lcd/dogm/dogm_Statusscreen.h:46:12: fatal error: ../../../_Statusscreen.h: No such file or directory
#include "../../../_Statusscreen.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/LPC1768/src/src/lcd/dogm/status_screen_DOGM.cpp.o] Error 1

#### Additional Information
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2945921/Archive.zip)

kAdonis, 2019-03-08:
You have custom Bootscreen and Statusscreen enabled, you have to copy these files to your new marlin sources
```
// Enable to show the bitmap in Marlin/_Bootscreen.h on startup.
#define SHOW_CUSTOM_BOOTSCREEN

// Enable to show the bitmap in Marlin/_Statusscreen.h on the status screen.
#define CUSTOM_STATUS_SCREEN_IMAGE
```

reloxx13, 2019-03-08:
or use sourcetree to "merge" from git.

sourcetree is freeware gui git client.

cinealfa, 2019-03-09:
ay caramba! How stupid of me

closed 2019-03-09 by cinealfa